espressif/esp-at #491: Rapid pings cause error

AlyoshaVasilieva, 2020-11-15:
## Environment

- Development Kit: ESP32-S2-Saola-1M
- Module or chip used: ESP32-S2-WROOM
- IDF version: v4.2-dev-2044-gdd3c032
- Build System: Using pre-built binary release 2.1.0.0
- Compiler version: Using pre-built binary release 2.1.0.0
- Operating System: Windows
- Power Supply: USB (have tested two different supply setups, same bug)

AT firmware `GMR` info:

```
AT version:2.1.0.0(0b76313 - ESP32S2 - Aug 20 2020 05:57:43)
SDK version:v4.2-dev-2044-gdd3c032
compile time(b5e1674):Aug 21 2020 04:59:57
Bin version:2.1.0(WROOM)
```

## Problem Description

Rapidly sending pings causes an error.

### Expected Behavior

I can ping a host rapidly with no issues.

### Actual Behavior

ESP32-S2 displays error over USB debug log and does not give `ERROR` or `OK` in AT UART. No longer responds to AT commands.

Rapid HTTP requests from a local server don't cause any issue.

### Steps to reproduce

1. Set up: SYSLOG=1, station mode, join AP
2. Send command `AT+PING="1.1.1.1"` as fast as possible (waiting until after `OK`, of course)

## Debug Logs

From hitting RST button again to error:

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x1988
load:0x40050000,len:0x1894
load:0x40054000,len:0x210c
entry 0x40050314
I (46) boot: ESP-IDF v4.2-dev-2044-gdd3c032 2nd stage bootloader
I (46) boot: compile time 04:59:46
I (46) boot: chip revision: 0
I (49) qio_mode: Enabling default flash chip QIO
I (54) boot.esp32s2: SPI Speed      : 80MHz
I (59) boot.esp32s2: SPI Mode       : QIO
I (64) boot.esp32s2: SPI Flash Size : 4MB
I (68) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (100) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (107) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (115) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (122) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (130) boot: End of partition table
I (134) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f000020 size=0x21338 (135992) map
I (168) esp_image: segment 1: paddr=0x00121360 vaddr=0x3ffc9360 size=0x0310c ( 12556) load
I (171) esp_image: segment 2: paddr=0x00124474 vaddr=0x40024000 size=0x00404 (  1028) load
I (175) esp_image: segment 3: paddr=0x00124880 vaddr=0x40024404 size=0x0b798 ( 47000) load
I (194) esp_image: segment 4: paddr=0x00130020 vaddr=0x40080020 size=0xae3e4 (713700) map
I (324) esp_image: segment 5: paddr=0x001de40c vaddr=0x4002fb9c size=0x097bc ( 38844) load
I (333) esp_image: segment 6: paddr=0x001e7bd0 vaddr=0x40070000 size=0x0001c (    28) load
I (345) boot: Loaded app from partition at offset 0x100000
I (345) boot: Disabling RNG early entropy source...
2.1.0
max tx power=78,ret=0
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=50 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 50ms
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=20 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 20ms
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=19 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 19ms
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=24 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 24ms
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=23 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 23ms
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=23 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 23ms
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=22 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 22ms
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=22 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 22ms
64 bytes from 1.1.1.1 icmp_seq=1 ttl=59 time=21 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 21ms
E (21482) ping_sock: esp_ping_new_session(246): create socket failed: -1
```

ustccw, 2021-01-12:
@AlyoshaVasilieva thanks for your report! we will reproduce it and fix.

ustccw, 2021-02-01:
@AlyoshaVasilieva we updated the esp-idf version for esp32-s2, and tested `AT+PING` command more than 7000 times, it looks like work well.

Please `git pull` the latest version and test, feel free to let me know if the issue still happens.

closed 2021-04-14 by FayeY