espressif/esp-at #492: esp-at是否可以使用到esp8266上呢

woshi520laoxie, 2020-11-24:
1、请问esp-at是否可以用到esp8266上
2、esp8266的AT指令是否以使用http的指令
3、是否有esp826的官方固件可以使用http指令的。

xcguang, 2020-11-25:
https://docs.espressif.com/projects/esp-at/en/latest/AT_Binary_Lists/index.html

woshi520laoxie, 2020-11-25:
你好，我下载了你提供的固件:  ESP8266-IDF-AT_V2.1
根据下载配置文件提示:
进行了下载，但是模组无法启动，请问这个问题是出在哪里呢？
--flash_mode dio 
--flash_freq 80m 
--flash_size 2MB

0x0 bootloader/bootloader.bin
0x8000 partition_table/partition-table.bin
0x9000 ota_data_initial.bin
0x10000 esp-at.bin

0xF0000 at_customize.bin
0xF1000 customized_partitions/factory_param.bin

0xF2000 customized_partitions/server_cert.bin
0xF4000 customized_partitions/server_key.bin
0xF6000 customized_partitions/server_ca.bin

0xF8000 customized_partitions/client_cert.bin
0xFA000 customized_partitions/client_key.bin
0xFC000 customized_partitions/client_ca.bin

0x104000 customized_partitions/mqtt_cert.bin
0x106000 customized_partitions/mqtt_key.bin
0x108000 customized_partitions/mqtt_ca.bin

下面的是我烧录的配置
[image]
下面的这个是我所使用的模组
[image]

vcmania, 2020-12-03:
hi.
my english is bad.
your image first line address 0 then bootloader.bin but partition-table.bin.
2MB(Mega Byte) is 16Mbit(Mega Bit).

xcguang, 2020-12-03:
请参考硬件连接图：https://docs.espressif.com/projects/esp-at/en/latest/Get_Started/Hardware_connection.html

woshi520laoxie, 2020-12-03:
@vcmania 
你好：
谢谢提醒，的确地址0的确应该是boot，我下载成分区表了，我晚些回去试试。

woshi520laoxie, 2020-12-03:
@xcguang @vcmania 你好，请问esp8266使用的AT指令，有http的指令吗？
我看介绍是esp32和esp8266都有的，《HTTP AT Commands    并没有显示是 ESP32 only》
但是我之前的固件还不带这个功能，目前哪个版本的固件是有http的AT指令功能的呢？

xcguang, 2020-12-04:
需要自己 在 menuconfig 配置编译

closed 2021-03-02 by xcguang