espressif/esp-at #493: Wakeup fail from light sleep

Honzik321, 2020-11-24:
Hello,

I found out new release ESP32-AT V2.1.0.0 20200728

In the Release Note, there is written: "SDK Update and Known issue: **Wakeup fail from light sleep**"

[image]

When this issue will be solved ? I waiting for almost one year, without light sleep my project cannot be completed.

Thank you,
Jan Kijonka,
HKMeas s.r.o. developer

xcguang, 2020-11-25:
Will you wakeup esp device by GPIO or UART?

Honzik321, 2020-11-25:
I tested both UART and then GPIO wake-up sources. The problem is always the same, after wake-up, the UART starts to repond, but the reponses are some unspecified characters (the UART communication never wake up correctly and never respond normally). I tested the ESP32-AT V2.1.0.0. 20200728 (older release has the same problem).

ArthurTimmermans, 2021-01-22:
I'm also having the same problem here.

Honzik321, 2021-01-22:
Hello, any progress in this issue ? I can offer monetary reward if this issue is solved in the following SDK release. Contact me on kijonka84@gmail.com, HKmeas s.r.o. company, Czech Rep.

Alson-tang, 2021-03-09:
@Honzik321 Hello, we are looking at the question. Thank you for your feedback.

ustccw, 2021-03-30:
@Honzik321 @ArthurTimmermans very sorry for the late reply.

It got fixed on https://github.com/espressif/esp-at/commit/0605b8fa7d55e1a9a261974c14f681453daa3777,  could you please `git pull` the latest code and test.

closed 2021-04-14 by FayeY