espressif/esp-at #494: AT+HTTPCLIENT error code 0x01030000

uilter, 2020-11-26:
Hi,

I'm facing an issue to send a large message using "AT+HTTPCLIENT".

My application send a JSON file over TCP, but I got an error code **"ERR CODE:0x01030000"** when I try to send a large command.

**Example:**
The command below works fine:
`AT+HTTPCLIENT=3,1,"http://192.168.15.182:3000/produtos",,,1,"{\"preco\":100\,\"produto\":\"barra\"\,\"array\":[{\"name\":\"H5G\"\,\"value\":-32}]\,\"alarms\":[{\"code\":2\,\"date\":1605133186}\,{\"code\":3\,\"date\":1605133126}]}"`

When I increase the size of JSON, it return the code "ERR CODE:0x01030000". Below is the command that I get the error.
`AT+HTTPCLIENT=3,1,"http://192.168.15.182:3000/produtos",,,1,"{\"preco\":100\,\"produto\":\"barra\"\,\"array\":[{\"name\":\"H5G\"\,\"value\":-32}\,{\"name\":\"H2G\"\,\"value\":10}]\,\"alarms\":[{\"code\":2\,\"date\":1605133186}\,{\"code\":3\,\"date\":1605133126}]}"`

I'm using the WROOM-32. Version about firmware is below.

> AT+GMR
> AT version:2.1.0.0(883f7f2 - Jul 24 2020 11:50:07)
> SDK version:v4.0.1-193-ge7ac221
> compile time(0ad6331):Jul 28 2020 02:47:21
> Bin version:2.1.0(WROOM-32)

Thank you!

uilter, 2020-11-27:
An update, using "AT+CIPSEND" I got no error and the data was sent without problem.

FayeY, 2020-12-23:
Hi,

The max length of an AT command is 256 bytes, I think the problem you met is due to the exceeded length.

We will add supporting HTTP POST long data into our development plan.

Sorry for the inconvenience, maybe you can use "AT+CIPSEND" instead for now.

Thanks.

uilter, 2021-01-18:
FayeY, using "AT+CIPSEND" works fine.

Thanks for confirm the issues.

closed 2021-01-18 by uilter