espressif/esp-at #497: esp32s2 SDIO AT build fail

lgl88911, 2021-01-03:
Follow doc to build WSP32S2 SDIO AT:
`./build menuconfig`
select: PLATFORM_ESP32S2，WROVER
config: Component config  --->AT  ---> communicate method for AT command, select"SDIO"

`./build build` will get link error, Did I miss any setup steps?

>/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj):(.literal.at_sdio_write_data+0x10): undefined reference to `sdio_slave_transmit'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj):(.literal.at_sdio_read_data+0xc): undefined reference to `sdio_slave_recv_load_buf'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj):(.literal.at_sdio_recv_task+0x8): undefined reference to `sdio_slave_recv'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj):(.literal.esp_at_sdio_slave_init+0x18): undefined reference to `sdio_slave_initialize'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj):(.literal.esp_at_sdio_slave_init+0x1c): undefined reference to `sdio_slave_recv_register_buf'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj):(.literal.esp_at_sdio_slave_init+0x20): undefined reference to `sdio_slave_set_host_intena'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj):(.literal.esp_at_sdio_slave_init+0x24): undefined reference to `sdio_slave_start'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj): in function `at_sdio_write_data':
/home/frank/esp/esp-at/build/../main/interface/sdio/at_sdio_task.c:90: undefined reference to `sdio_slave_transmit'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj): in function `at_sdio_read_data':
/home/frank/esp/esp-at/build/../main/interface/sdio/at_sdio_task.c:144: undefined reference to `sdio_slave_recv_load_buf'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj): in function `at_sdio_recv_task':
/home/frank/esp/esp-at/build/../main/interface/sdio/at_sdio_task.c:192: undefined reference to `sdio_slave_recv'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(at_sdio_task.c.obj): in function `esp_at_sdio_slave_init':
/home/frank/esp/esp-at/build/../main/interface/sdio/at_sdio_task.c:162: undefined reference to `sdio_slave_initialize'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/frank/esp/esp-at/build/../main/interface/sdio/at_sdio_task.c:167: undefined reference to `sdio_slave_recv_register_buf'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/frank/esp/esp-at/build/../main/interface/sdio/at_sdio_task.c:168: undefined reference to `sdio_slave_recv_load_buf'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/frank/esp/esp-at/build/../main/interface/sdio/at_sdio_task.c:166: undefined reference to `sdio_slave_set_host_intena'
/home/frank/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/frank/esp/esp-at/build/../main/interface/sdio/at_sdio_task.c:174: undefined reference to `sdio_slave_start'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

lgl88911, 2021-01-03:
Modify as below, Link Pass， But I want to find out whether esp32s supports SDIO AT? If not, maybe I have to return esp32-s2 and repurchase esp32.
esp-at\esp-idf\components\driver\CMakeLists.txt的esp32s2 target add source "sdio_slave.c"
esp-at\esp-idf\components\soc\soc\esp32s2\include\soc\periph_defs.h, enum periph_module_t add PERIPH_SDIO_SLAVE_MODULE

ustccw, 2021-01-12:
@lgl88911 Hi, SDIO is not supported on ESP32-S2 hardware. 

please try with ESP32 chip.

closed 2021-01-12 by igrr

lgl88911, 2021-01-18:
@ustccw Thanks, ESP32 is one of the choices, we prefer to use ESP32-S2.
 SPI AT on S2 also have link error, Is there any plan to support SPI AT on S2?

xcguang, 2021-01-18:
@ESP-Coco please help to have a look

ESP-Coco, 2021-01-19:
@lgl88911 Hi，we are developing support for ESP32S2 SPI AT, it will be released soon.

lgl88911, 2021-01-19:
@xcguang  @ESP-Coco Thanks for reply, Waiting for your good news！