espressif/esp-at #498: WPA3 / WPA2 transistion mode PMF support on WROOM-32

ubichsel, 2021-01-06:
Dear Espressif Team

I want to connect to a WPA3 WIFI network and it is not working at the moment.
With WPA2-WPA3 transition mode enabled on router (and PMF set to force) it will not work anymore. So my guess is, the PMF flags are not set in wifi_sta_config_t->pmf_cfg. Is there an API where I can set those parameters or is it a bug / just not implemented yet in AT-Firmware? In WPA-3 PMF is mandatory to use.

When PMF is set to forced on router I get the following scan result (WPA2 / WPA3 transition mode):
+CWLAP:(**0**,"WR03-Synology",-48,"00:11:32:AB:5D:41",5)

When PMF is set to optional I get the following scan result (correct) and I can connect over WPA2:
+CWLAP:(**7**,"WR03-Synology",-51,"00:11:32:AB:5D:41",5)

The router I try it out is: Synology MR2200ac

ESP Device: WROMM-32
AT+GMR:
AT version:2.2.0.0-dev(75f973a - ESP32 - Jan  4 2021 10:05:25)
SDK version:v4.0.1-193-ge7ac221b4
compile time(0e33df1):Jan  5 2021 08:47:03
Bin version:00.25(WROOM-32)

Note: The WPA3 support is activated in menuconfig

When PMF is working in transition mode I can go on and test if WPA3-PSK is working as well.

Thank you for your help.

ustccw, 2021-01-11:
@ubichsel this feature will sync to GitHub this week, while WPA3 has already merged internally, and do some test cases.

ubichsel, 2021-01-11:
@ustccw Thanks for the update, I'm looking forward for the update and let you know if it works

closed 2021-01-15 by igrr

ustccw, 2021-01-15:
@ubichsel please `git pull` latest version and test https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/Wi-Fi_AT_Commands.html#at-cwjap-connect-to-an-ap.

please feel free to let me know if anything not clear.

ranganath-grl, 2022-02-16:
WPA3 WIFI network and it is not working at the moment when we use power save mode menuconfig. Is there any solution? 
Using latest version