espressif/esp-at #503: Build issue (with Ethernet)

ubichsel, 2021-01-20:
Hi Espressif team

I updated to newest version (esp-at) on my windows machine and now I have several problems when compiling (WROOM-32):

-The include folder in main is not in compiler include directories. Get error:
`esp-at/main/app_main.c:44:10: fatal error: at_interface.h: No such file or directory`
 `include "at_interface.h"`

`esp-at/main/interface/uart/at_uart_task.c:39:10: fatal error: at_interface.h: No such file or directory`
 `#include "at_interface.h"`
For all interfaces in interface folder

As a temporary workaround I copy the file at_interface.h from main/include to main folder. Maybe you can correct this issue somewhen.

I need Ethernet support, in earlier version I overwrote `esp_at_get_eth_default_config` function to have my pins initialized correctly. Now I get the error:
`libesp32_at_core.a(at_ethCmd.c.obj): in function `esp_at_eth_cmd_regist':`
`(.text.esp_at_eth_cmd_regist+0x6): undefined reference to `esp_at_get_eth_default_config`

I guess during the refactoring some parts where not correctly moved with all references / includes.

Thank you for your help.

ustccw, 2021-02-09:
@ubichsel please `git pull` the latest version and test again.

If the issue still exist, please provide the esp-at version by `git describe --tags`, and a complete compilation log.

ubichsel, 2021-02-23:
Dear @ustccw, thanks for answering me
I tried to build a gain but it was the same error. I have a windows PC, downloaded the newest msys32 package and cloned the most recent ESP-AT project.
git describe --tags => v2.1.0.0_esp32s2-247-g8e83329

Let's start with the first error. Probably when this one is solved, others are solved too:

kconfiglib.KconfigError: main/Kconfig:12: '$ESP_AT_PORJECT_PATH/main/interface/uart/Kconfig' not found (in 'source "$ESP_AT_PORJECT_PATH/main/interface/uart/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
make: *** No rule to make target '/home/myName/esp-neu/v0025/esp-at/build/include/config/auto.conf', needed by '/home/myName/esp-neu/v0025/esp-at/build/bootloader/bootloader.bin'.  Stop.
[compiler_logs.txt](https://github.com/espressif/esp-at/files/6027315/compiler_logs.txt)

Is there something not documented or special which I have to do for the variable: $ESP_AT_PORJECT_PATH in file main/Kconfig? I'm not really a python or KCONFIG expert so forgive me if it is a stupid question. When I replace the $ESP_AT_PORJECT_PATH with absolute path from files the error disappears.

ustccw, 2021-02-23:
@ubichsel esp-at now uses the cmake instead of make, please build the project according to 
https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html

```
rm build sdkconfig -rf
./build.py build
```

ubichsel, 2021-02-26:
Dear @ustccw,
Thank you for the response. Indeed, I used to old build chain from earlier times. With the new one (cmake) the build is working. Unfortunately I get another problem with my custom partition csv files (same as used before with earlier version of AT-Stack). When I set my own at_customize.csv and partitions_at.csv (attached) compilation and download works but then the watchdog is triggered after calling the initial BLE commands:
AT+BLEINIT=2
 => OK
AT+BLEGATTSSRVCRE
Stucked here (no response for any AT-Commands)

**Monitor console:**
```Backtrace:0x400D43CD:0x3FFBE370 0x4008277E:0x3FFBE390 0x400DBBCF:0x3FFD5380 0x400DBC44:0x3FFD53C0 0x400DF4B6:0x3FFD53E0 0x400E9AF9:0x3FFD5400 0x4008F402:0x3FFD5440
0x400d43cd: task_wdt_isr at C:/msys32/home/ueli.bichsel/esp-neu/v0025/esp-at/esp-idf/components/esp_common/src/task_wdt.c:187

0x4008277e: _xt_lowint1 at C:/msys32/home/ueli.bichsel/esp-neu/v0025/esp-at/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x400dbbcf: at_setupCmdBLeSacnRspData at ??:?

0x400dbc44: at_exeCmdBlesCreSrv at ??:?

0x400df4b6: at_cmd_process at ??:?

0x400e9af9: at_handle_result_code at ??:?

0x4008f402: vPortTaskWrapper at C:/msys32/home/ueli.bichsel/esp-neu/v0025/esp-at/esp-idf/components/freertos/xtensa/port.c:143
```

If I use the standard at_customize.csv everything is working. I need my own at_customize.csv to be compatible with existing products. Can you please have a look why this error happens?

Thank you again for the support.
[partitions.zip](https://github.com/espressif/esp-at/files/6050106/partitions.zip)

ustccw, 2021-03-01:
@ubichsel could you please raise another issue for new?

ubichsel, 2021-03-01:
Sorry, I made a new one (#514 ) and close this for now

closed 2021-03-01 by ubichsel