espressif/esp-at #504: How to use `\<any>` feature?

murphyzhao, 2021-01-24:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

```
\<any>: escape <any> character means that drop backslash symbol and only use <any> character
```

How to use [`\<any>`](https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/index.html) feature?

ustccw, 2021-02-09:
@murphyzhao this feature is suitable for special user scenarios.

please see as https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/TCP-IP_AT_Commands.html#cmd-sendex.

murphyzhao, 2021-02-13:
Get, thanks

closed 2021-02-13 by murphyzhao