espressif/esp-at #509: Cannot use vendor descriptor for BLE GATT

rfrowe, 2021-02-18:
## Environment
- Development Kit:  Macchina SuperB
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v4.2
- Build System: build.py
- Compiler version: 8.4.0
- Operating System: macOS
- Power Supply: USB

## Problem Description

I've been trying to set up my own BLE GATT service. I've managed to work my way through the existing documentation with some effort and get things working (almost). I'm able to successfully use my custom service and characteristics, but for some reason my custom descriptor is not showing.

Here is my BLE configuration:
```
index,uuid_len,uuid,perm,val_max_len,val_cur_len,value
0,16,0x2800,0x1,2,2,180A
1,16,0x2803,0x1,1,1,2
2,16,0x2A29,0x1,16,10,5865766f2c20496e632e
3,16,0x2803,0x1,1,1,2
4,16,0x2A26,0x1,16,6,76302e302e31
5,16,0x2800,0x1,16,16,338586FB4CAB422A87C479CFD0B2D2D2
6,16,0x2803,0x1,1,1,10
7,128,0x338586FC4CAB422A87C479CFD0B2D2D2,0x1,9,0,
8,128,0x338586FE4CAB422A87C479CFD0B2D2D2,0x1,1,1,0
9,16,0x2902,0x11,2,2,0000
10,16,0x2803,0x1,1,1,10
11,128,0x338586FC4CAB422A87C479CFD0B2D2D2,0x1,9,0,
12,128,0x338586FE4CAB422A87C479CFD0B2D2D2,0x1,1,1,0
13,16,0x2902,0x11,2,2,0000
14,16,0x2803,0x1,1,1,10
15,128,0x338586FD4CAB422A87C479CFD0B2D2D2,0x1,9,0,
16,128,0x338586FE4CAB422A87C479CFD0B2D2D2,0x1,1,1,0
17,16,0x2902,0x11,2,2,0000
```

Above, we can see that all of the descriptors specified in the CSV: three `0x2902` (CCCD) and three `0x338586FE4CAB422A87C479CFD0B2D2D2` (my custom descriptor). 

### Expected Behavior

```
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0x2A29,0x02
+BLEGATTSCHAR:"char",1,2,0x2A26,0x02
+BLEGATTSCHAR:"char",2,1,0x338586FC4CAB422A87C479CFD0B2D2D2,0x10
+BLEGATTSCHAR:"desc",2,1,1,0x338586FE4CAB422A87C479CFD0B2D2D2
+BLEGATTSCHAR:"desc",2,1,2,0x2902
+BLEGATTSCHAR:"char",2,2,0x338586FC4CAB422A87C479CFD0B2D2D2,0x10
+BLEGATTSCHAR:"desc",2,2,1,0x338586FE4CAB422A87C479CFD0B2D2D2
+BLEGATTSCHAR:"desc",2,2,2,0x2902
+BLEGATTSCHAR:"char",2,3,0x338586FD4CAB422A87C479CFD0B2D2D2,0x10
+BLEGATTSCHAR:"desc",2,2,1,0x338586FE4CAB422A87C479CFD0B2D2D2
+BLEGATTSCHAR:"desc",2,3,2,0x2902
```

### Actual Behavior

```
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0x2A29,0x02
+BLEGATTSCHAR:"char",1,2,0x2A26,0x02
+BLEGATTSCHAR:"char",2,1,0x338586FC4CAB422A87C479CFD0B2D2D2,0x10
+BLEGATTSCHAR:"char",2,2,0x338586FC4CAB422A87C479CFD0B2D2D2,0x10
+BLEGATTSCHAR:"char",2,3,0x338586FD4CAB422A87C479CFD0B2D2D2,0x10
```

When I connect to the device via an Android Tablet, I'm able to see all three characteristics and the CCCD for each (despite it not being listed in the output of the above AT command). However, my custom descriptor is not present.

If I remove my custom descriptors from the CSV but leave the CCCD descriptors, I see an improvement:

```
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0x2A29,0x02
+BLEGATTSCHAR:"char",1,2,0x2A26,0x02
+BLEGATTSCHAR:"char",2,1,0x338586FC4CAB422A87C479CFD0B2D2D2,0x10
+BLEGATTSCHAR:"desc",2,1,1,0x2902
+BLEGATTSCHAR:"char",2,2,0x338586FC4CAB422A87C479CFD0B2D2D2,0x10
+BLEGATTSCHAR:"desc",2,2,1,0x2902
+BLEGATTSCHAR:"char",2,3,0x338586FD4CAB422A87C479CFD0B2D2D2,0x10
+BLEGATTSCHAR:"desc",2,3,1,0x2902
```

I also tried changing the first instance of my custom descriptor to `0x2901` (Characteristic User Description). When I do this, `AT=BLEGATTSCHAR?` reports `0x2901` and `0x2902` descriptors on the first characteristic, then none for the rest.

This would seem to indicate to me that something is going wrong while processing my custom descriptors. It seems processing my custom descriptor also prevents it from properly processing subsequent SIG descriptors correctly (though not entirely, since the client still reports the, it's just that the server won't show them via the AT command). Are vendor descriptors not supported?

rfrowe, 2021-02-19:
Ok I tried using a descriptor UUID of `0x2910` which is not assigned in the Bluetooth SIG [list](https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf). This succeeds. My best guess right now is there's a failure specifically for 128-bit characteristic UUIDs

rfrowe, 2021-02-19:
Actually upon further investigation, although `AT+BLEGATTSCHAR?` shows the `0x2910` descriptor, none of my BLE clients can see it on the characteristic.

rfrowe, 2021-03-03:
Does anyone have any info on support for 128-bit descriptor UUIDs?

Luxin812, 2021-09-14:
Regarding how AT customizes the BLE service, you can refer to this document: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_customize_BLE_services.html. Hope it helps you.

rfrowe, 2021-10-05:
@Luxin812 That document is the one I had been referencing and it contains no information on 128-bit vendor descriptors. I ended up making a post on the official ESP32 forum (https://www.esp32.com/viewtopic.php?f=2&t=19532) back in February and have never received a responsee.

Luxin812, 2022-04-19:
@rfrowe For documentation on supporting 128-bit descriptor UUIDs, you can refer to: https://www.bluetooth.com/specifications/assigned-numbers/