espressif/esp-at #510: ESP32-WROOM-32D with Poor Signal Strengths

SamuelHenrique32, 2021-02-24:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32D
- Build System: Using pre-built binary release 2.1.0.0
- Compiler version: Using pre-built binary release 2.1.0.0
- Power Supply: external 3.3V

## Problem Description

Chip has signal instability in distances bigger than 10m from Wi-Fi access point.

### Expected Behavior

A great signal strengthness in measured distances from 10m to 30m according to datasheet.

### Actual Behavior

RSSI is varying and the connection to the access point is poor. Furthermore, when pinging the chip there are a lot of packet misses.

### Steps to reproduce

The following AT Commands were sent to the chip:

1. Set network SSID with "AT+CWJAP="
2. Set mode with "AT+CIPMUX=" as Soft AP + Station Mode
3. Enable DHCP
4. Get RSSI with "AT+CWJAP?\r\n"

The test were conducted without any barrier between ESP32 and the access point.

I tried another ways to improve the signal quality (without success):

1. Verify RF Power with command "AT+RFPOWER?\r\n": chip answers with "+RFPOWER:78,2,2,2" indicating maximum signal performance
2. Change between protocol standard of station mode with (AT+CWSTAPROTO): resulted in very similar results for all the options

ustccw, 2021-02-26:
@SamuelHenrique32 The signal strength is greatly affected by the environment.

Could you please test ping function by esp-idf demo:
https://github.com/espressif/esp-idf/tree/master/examples/protocols/icmp_echo, and compare with the esp-idf latest firmware.
> you can download latest esp-at firmwae from GitHub Actions: https://github.com/espressif/esp-at/actions -> Click the latest workflow -> Click Artifacts

FayeY, 2021-04-14:
Closing due to lack of response. Please feel free to reopen if the issue persists and you can supply more details. Thanks.

closed 2021-04-14 by FayeY

gfanini, 2025-08-21:
can esp-idf ide generate a firmware for let's say older esp32 wroom 32D model of about 2020 ?
is it possible to configure it to receive AT commands on uart 2 pins 16-17 ? and would that uart enabled coexist say with usb interface that is on modules commonly available ?