espressif/esp-at #512: Compilation failed under windows

wajlh, 2021-02-25:
Compiling esp-at fails under windows,
How to solve it?
---------------------------------------
PS E:\ESP32> cd esp-at
PS E:\ESP32\esp-at> idf.py build
Executing action: all (aliases: build)
Running cmake in directory e:\esp32\esp-at\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 e:\esp32\esp-at"...
CMake Error at CMakeLists.txt:9 (string):
  string no output variable specified

CMake Error at CMakeLists.txt:13 (string):
  string begin index: 9 is out of range 0 - 0

CMake Error at CMakeLists.txt:14 (string):
  string sub-command STRIP requires two arguments.

CMake Error at CMakeLists.txt:15 (string):
  string no output variable specified

silence:0 sdkconfig:E:/ESP32/esp-at/module_config/module__default/sdkconfig.defaults
CMake Error at CMakeLists.txt:49 (include):
  include could not find load file:

    E:/ESP32/esp-at/esp-idf/tools/cmake/project.cmake

-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:51 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:51 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "E:/ESP32/esp-at/build/CMakeFiles/CMakeOutput.log".
See also "E:/ESP32/esp-at/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
PS E:\ESP32\esp-at>

ustccw, 2021-02-26:
@wajlh please build the esp-at project according to https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html

wajlh, 2021-02-28:
@ustccw thank you for your reply .
I used "ESP-IDF Tools Installer" to install the windows environment and executed
step 1: install python>=3.8
step 2: [install pip](https://pip.pypa.io/en/latest/installing/)
step 3: install the following python packages with pip: python -m pip install pyyaml xlrd
The test "esp-idf\examples\ethernet\eth2ap" project compiles normally. But the ESP-AT project cannot be compiled normally, and the log is the same as before.
Do I have any other issues that I need to pay attention to?

ustccw, 2021-03-01:
@wajlh esp-at uses `./build.py build`, other than `idf.py build`. please try again and provide the compile log.

riccardobritti82, 2021-03-11:
> ## Compiling esp-at fails under windows,
> How to solve it?
> PS E:\ESP32> cd esp-at
> PS E:\ESP32\esp-at> idf.py build
> Executing action: all (aliases: build)
> Running cmake in directory [...]

I have exactly the same problem.
I think the problem is about the IDF_PATH but I don't undestand where I have the problem.
the command :

build.py menuconfig     work fine and i can set the platform etc...

but when I call :

build.py build the log is:

D:\Progetti\ESP32\Progetti_ESP\Progetti_ESP\ESP_AT_from_git\esp-at>build.py build
module_name WROOM-32
platform_name=ESP32,module_name=WROOM-32
WARNING: IDF_PATH environment variable is set to d:\progetti\esp32\progetti_esp\idf_path but esp-idf\tools\idf.py path indicates IDF directory d:\progetti\esp32\progetti_esp\progetti_esp\esp_at_from_git\esp-at\esp-idf. Using the environment variable directory, but results may be unexpected...
Usage: esp-idf\tools\idf.py [OPTIONS] COMMAND1 [ARGS]... [COMMAND2 [ARGS]...]...

  ESP-IDF CLI build management tool. For commands that are not known to idf.py an attempt to execute it as a build
  system target will be made.

Options:
  --version                       Show IDF version and exit.
  --list-targets                  Print list of supported targets and exit.
  -C, --project-dir PATH          Project directory.
  -B, --build-dir PATH            Build directory.
  -w, --cmake-warn-uninitialized / -n, --no-warnings
                                  Enable CMake uninitialized variable warnings for CMake files inside the project
                                  directory. (--no-warnings is now the default, and doesn't need to be specified.)

  -v, --verbose                   Verbose build output.
  --ccache / --no-ccache          Use ccache in build. Disabled by default, unless IDF_CCACHE_ENABLE environment
                                  variable is set to a non-zero value.

  -G, --generator [Ninja|MinGW Makefiles]
                                  CMake generator.
  -D, --define-cache-entry TEXT   Create a cmake cache entry. This option can be used at most once either globally, or
                                  for one subcommand.

  -b, --baud INTEGER              Baud rate for flashing. The default value can be set with the ESPBAUD environment
                                  variable. This option can be used at most once either globally, or for one
                                  subcommand.

  -p, --port TEXT                 Serial port. The default value can be set with the ESPPORT environment variable.
                                  This option can be used at most once either globally, or for one subcommand.

  --help                          Show this message and exit.

Commands:
  all                    Aliases: build. Build the project.
  app                    Build only the app.
  app-flash              Flash the app only.
  bootloader             Build only bootloader.
  bootloader-flash       Flash bootloader only.
  clean                  Delete build output files from the build directory.
  confserver             Run JSON configuration server.
  efuse_common_table     Generate C-source for IDF's eFuse fields.
  efuse_custom_table     Generate C-source for user's eFuse fields.
  encrypted-app-flash    Flash the encrypted app only.
  encrypted-flash        Flash the encrypted project.
  erase_flash            Erase entire flash chip.
  erase_otadata          Erase otadata partition.
  flash                  Flash the project.
  fullclean              Delete the entire build directory contents.
  gdb                    Run the GDB.
  gdbgui                 GDB UI in default browser.
  gdbtui                 GDB TUI mode.
  menuconfig             Run "menuconfig" project configuration tool.
  monitor                Display serial output.
  openocd                Run openocd from current path
  partition_table        Build only partition table.
  partition_table-flash  Flash partition table only.
  post_debug             Utility target to read the output of async debug action and stop them.
python-clean           Delete generated Python byte code from the IDF directory
  read_otadata           Read otadata partition.
  reconfigure            Re-run CMake.
  set-target             Set the chip target to build.
  show_efuse_table       Print eFuse table.
  size                   Print basic size information about the app.
  size-components        Print per-component size information.
  size-files             Print per-source-file size information.
idf.py build ret: 0
[Errno 2] No such file or directory: 'build\\flash_project_args'

D:\Progetti\ESP32\Progetti_ESP\Progetti_ESP\ESP_AT_from_git\esp-at>

Can you help me ?

wajlh, 2021-03-15:
> > ## 在Windows下，编译esp-at失败，
> > 怎么解决呢？
> > PS E：\ ESP32> cd esp-at 
> > PS E：\ ESP32 \ esp-at> idf.py build
> > 执行操作：全部（别名：build）
> > 在目录e：\ esp32 \ esp-at \ build中
> > 运行cmake执行“ cmake -G Ninja -DPYTHON_DEPS_CHECKED = 1 -DESP_PLATFORM = 1 -DCCACHE_ENABLE = 1 e：\ esp32 \ esp-at“ ... 
> > CMakeLists.txt：9处的CMake错误：（字符串）：
> > 字符串未指定输出变量
> > CMakeLists.txt：13（字符串）的CMake错误：
> > 字符串开始索引：9超出范围0-0
> > CMakeLists.txt：14（字符串）处的CMake错误：
> > 字符串子命令STRIP需要两个参数。
> > CMakeLists.txt处的CMake错误：15（字符串）：
> > 字符串未指定输出变量
> > 静默：0 sdkconfig：E：/ESP32/esp-at/module_config/module__default/sdkconfig.defaults 
> > CMakeLists.txt：49上的CMake错误（包含）：
> > 包含找不到加载文件：
> > ```
> > E:/ESP32/esp-at/esp-idf/tools/cmake/project.cmake
> > ```
> > 
> > 
> > -C编译器标识未知
> > -CXX编译器标识未知
> > CMakeLists.txt：51（项目）处的CMake错误：
> > 找不到CMAKE_C_COMPILER。
> > 通过将环境
> > 变量“ CC”或CMake缓存条目CMAKE_C_COMPILER设置
> > 为编译器的完整路径，或将其设置为PATH中的编译器名称，来告诉CMake在哪里找到编译器。
> > CMakeLists.txt：51（项目）处的CMake错误：
> > 找不到CMAKE_CXX_COMPILER。
> > 通过将环境
> > 变量“ CXX”或CMake缓存条目CMAKE_CXX_COMPILER
> > 设置为编译器的完整路径，或将其设置为PATH中的编译器名称，来告诉CMake在哪里找到编译器。
> > -配置不完整，发生错误！
> > 另请参见“ E：/ESP32/esp-at/build/CMakeFiles/CMakeOutput.log”。
> > 另请参见“ E：/ESP32/esp-at/build/CMakeFiles/CMakeError.log”。
> > cmake失败，退出代码为1 
> > PS E：\ ESP32 \ esp-at>
> 
> 我也有完全一样的问题。
> 我认为问题出在IDF_PATH上，但我不理解出现问题的地方。
> 命令 ：
> 
> build.py menuconfig工作正常，我可以设置平台等...
> 
> 但是当我打电话时：
> 
> build.py构建日志为：
> 
> D：\ Progetti \ ESP32 \ Progetti_ESP \ Progetti_ESP \ ESP_AT_from_git \ esp-at> build.py menuconfig
> module_name WROOM-32
> platform_name = ESP32，module_name = WROOM-32
> 用法：esp-idf \ tools \ idf.py [OPTIONS] COMMAND1 [ ARGS] ... [COMMAND2 [ARGS] ...] ...
> 
> ESP-IDF CLI构建管理工具。对于idf.py未知的命令，将尝试将其作为构建系统目标执行
> 。
> 
> 选项：
> --version显示IDF版本并退出。
> --list-targets打印支持的目标列表并退出。
> -C，--project-dir PATH项目目录。
> -B，--build-dir PATH构建目录。
> -w，--cmake-warn-uninitialized / -n，--no-
> warnings在项目目录中为CMake文件启用CMake未初始化变量警告。（--no-
> 警告现在是默认值，不需要指定。）
> 
> -v，--verbose详细的构建输出。
> --ccache / --no-ccache在构建中使用ccache。默认情况下禁用，除非将IDF_CCACHE_ENABLE环境变量设置为
> 非零值。
> 
> -G，--generator [Ninja | MinGW Makefiles]
> CMake生成器。
> -D，--define-cache-entry TEXT创建cmake缓存条目。此选项最多可以全局使用一次，也可以用于一个子命令。
> -b，--baud INTEGER闪烁的波特率。可以使用ESPBAUD环境变量设置默认值。此选项
> 最多可以全局使用一次，也可以用于一个子命令。
> 
> -p，--port TEXT串行端口。可以使用ESPPORT环境变量设置默认值。此选项
> 最多可以全局使用一次，也可以用于一个子命令。
> 
> --help显示此消息并退出。
> 
> 命令：
> 所有别名：构建。生成项目。
> app仅构建应用程序。
> app-flash仅刷新应用程序。
> 引导加载程序仅构建引导加载程序。
> bootloader-flash仅Flash引导加载程序。
> clean从构建目录中删除构建输出文件。
> confserver运行JSON配置服务器。
> efuse_common_table为IDF的eFuse字段生成C源代码。
> efuse_custom_table为用户的eFuse字段生成C源代码。
> crypto-app-flash仅闪存加密的应用程序。
> 加密闪存闪存加密的项目。
> delete_flash擦除整个闪存芯片。
> delete_otadata擦除otadata分区。
> flash刷新项目。
> fullclean删除整个构建目录的内容。
> gdb运行GDB。
> gdbgui默认浏览器中的GDB UI。
> gdbtui GDB TUI模式。
> menuconfig运行“ menuconfig”项目配置工具。
> 监视器显示串行输出。
> openocd从当前路径运行openocd
> partition_table仅构建分区表。
> partition_table-flash仅Flash分区表。
> post_debug实用程序目标，用于读取异步调试操作的输出并停止它们。
> python-clean从IDF目录中删除生成的Python字节代码
> read_otadata读取otadata分区。
> 重新配置重新运行CMake。
> set-target设置要构建的芯片目标。
> show_efuse_table打印eFuse表。
> 大小打印有关应用程序的基本大小信息。
> size-components打印每个组件的尺寸信息。
> size-files打印每个源文件的大小信息。
> idf.py build ret：0
> [Errno 2]没有这样的文件或目录：'build \ flash_project_args'
> 
> 你能帮助我吗 ？

wajlh, 2021-03-21:
@riccardobritti82 I gave up the windows OS method,because using"./build.py build" method was also unsuccessful.  and used Ubuntu OS is ok

riccardobritti82, 2021-03-21:
I have solved the problem on windows... The guide is not clear, but now with the correct sequence, it work correctly
Thanks

MaJerle, 2021-04-01:
@riccardobritti82 can you share the sequence?

moltonim, 2021-07-20:
@riccardobritti82 please share the sequence-
Dai Riccardo condividi!!!

ustccw, 2022-05-27:
@wajlh @riccardobritti82 docs updated. esp-at release/v2.4.0.0 and master branch simplfied the compile workflow. just run `python build.py install` and `python build.py build` to build the esp-at project, could you please have a try?
see this doc for more details:
https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html