espressif/esp-at #515: AT COMMAND maximum socket number config.

jerrytry1, 2021-03-01:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|
- Kit versionv2
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.0.1-193-ge7ac221b4
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [LinuX]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
Hi,

1. What is the maximum number of socket that I can config. in the document it is 5.
Reference:
config AT_SOCKET_MAX_CONN_NUM
int "AT socket maximum connection number"

2. if I config the maximum number of socket to 16. will it affect link ID. (Will link ID also be increased?)
<link ID>: ID of the connection (0~4), used for multiple connections.

In our project we need at least 7 sockets open.  but The official release, only mentioned, linkID is (0~4)

Thanks

ustccw, 2021-03-05:
@jerrytry1 Hi, the maximum number of socket is limited to the minimum value of lwip sockets config and at sockets config. config them at:
- `./build.py menuconfig` -> `Component config` -> `LWIP` ->  `Max number of open sockets`
- `./build.py menuconfig` -> `Component config` -> `AT` ->  `AT socket maximum connection number`

BRs,
CW

FayeY, 2021-04-14:
Closing as this question appears to be answered. Please feel free to reopen if the issue persists and you can supply more details. Thanks.

closed 2021-04-14 by FayeY