espressif/esp-at #516: ESP32-WROOM-32E - Secure Simple Pairing deactivated doesn't build!

RMOX97, 2021-03-03:
### Hardware:
Board:							ESP32-WROOM-32E
AT version:			                      2.2.0.0	
SDK version:				                  4.2	
Flash Frequency:					40Mhz 
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I would like to enable pairing pin in Classic Bluetooth on ESP32.

After enabling the AT commands for bluetooth (via menuconfig) I would like to disable the "Secure Simple Pairing" in order to activate the PIN request for the bluetooth connection of a device.
the commands executed are the following:

_Component config --> AT --> [*] AT bt command support
Component config --> Bluetooth -->  Bluedroid Options --> [ ] Secure Simple Paring_

After saving the builder fails. I attach below what it returns.

### Sketch
[SecureSimpleParing_ERRORI.txt](https://github.com/espressif/arduino-esp32/files/6075567/SecureSimpleParing_ERRORI.txt)

ustccw, 2021-03-12:
@xiewenxiang please take a look

Luxin812, 2022-04-20:
Maybe you can test with the latest version.

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

closed 2023-06-05 by ustccw

bschousek, 2024-04-22:
I'm still seeing the error with release v4.0.0.0
[idf_py_stdout_output_22012.txt](https://github.com/espressif/esp-at/files/15064096/idf_py_stdout_output_22012.txt)