espressif/esp-at #517: ESP-AT Master compile error

jack685, 2021-03-04:
/mnt/hgfs/esp8266-share/esp-at/components/at/lib/libesp8266_at_core.a(at_pingCmd.c.obj):(.literal.esp_at_ping_process+0xc): undefined reference to `ping_init'
/mnt/hgfs/esp8266-share/esp-at/components/at/lib/libesp8266_at_core.a(at_pingCmd.c.obj): In function `esp_at_ping_process':
(.text.esp_at_ping_process+0x78): undefined reference to `ping_init'
collect2: error: ld returned 1 exit status
make: *** [/mnt/hgfs/esp8266-share/esp-at/esp-idf/make/project.mk:510: /mnt/hgfs/esp8266-share/esp-at/build/esp-at.elf] Error 1
root@ubuntu:/mnt/hgfs/esp8266-share/esp-at# 
root@ubuntu:/mnt/hgfs/esp8266-share/esp-at# git branch
* master

ustccw, 2021-03-05:
@jack685 could you please provide detailed compile log?

jack685, 2021-03-11:
hello ustcct, 
My platform choose PLATFORM_ESP8266 and module_name choose ESP8266_1MB. esp-at checkout to Master version. export then make menuconfig and log as follow.
root@ubuntu:/mnt/esp8285/esp-at# make menuconfig
Python requirements from /mnt/hgfs/esp8266-share/esp-at/esp-idf/requirements.txt are satisfied.
MENUCONFIG
Traceback (most recent call last):
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1128, in _open
    return open(filename)
FileNotFoundError: [Errno 2] No such file or directory: '$ESP_AT_PORJECT_PATH/main/interface/uart/Kconfig'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1158, in _enter_file
    self._file = self._open(filename)
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1141, in _open
    raise IOError(
OSError: Could not open '$ESP_AT_PORJECT_PATH/main/interface/uart/Kconfig' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/confgen.py", line 553, in <module>
    main()
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/confgen.py", line 225, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 653, in __init__
    self._parse_block(None,           # end_token
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1632, in _parse_block
    self._parse_block(_T_ENDMENU,
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1584, in _parse_block
    prev_node = self._parse_block(None,            # end_token
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1632, in _parse_block
    self._parse_block(_T_ENDMENU,
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1681, in _parse_block
    self._parse_block(_T_ENDCHOICE,
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1583, in _parse_block
    self._enter_file(sourced_file)
  File "/mnt/hgfs/esp8266-share/esp-at/esp-idf/tools/kconfig_new/kconfiglib.py", line 1167, in _enter_file
    .format(self._filename, self._linenr, e.message))
AttributeError: 'OSError' object has no attribute 'message'
make: *** No rule to make target 'menuconfig', needed by '/mnt/esp8285/esp-at/build/include/config/auto.conf'.  Stop.
root@ubuntu:/mnt/esp8285/esp-at#

jack685, 2021-03-11:
but I compile it sucessfull when i checkout  to release/v2.1.0.0_esp8266 .

ustccw, 2021-03-12:
@jack685 please build the esp-at project according to https://github.com/espressif/esp-at/blob/master/docs/en/Compile_and_Develop/How_to_clone_project_and_compile_it.rst

Moreover, it it recommended that keep the default sdk version.

FayeY, 2021-04-14:
Closing as this question appears to be answered. Please reopen if the issue persists and you can supply more details. Thanks.

closed 2021-04-14 by FayeY