espressif/esp-at #518: Port of esp-at to ESP32-C3

mhummels, 2021-03-05:
Hi,

I cannot find any at firmware for the new ESP32-C3 SoC. Please port this repo to the ESP32-C3 also, since it is almost a drop in replacement for the old ESP8266/ESP8285 when changing the pinout accordingly.

BRs,
mhummels

ustccw, 2021-03-05:
@mhummels Thanks for your observation.
ESP32-C3 had ported to esp-at project internally, and in testing now.  
Should be pushed to GitHub in the next two weeks

BRs,
CW

ustccw, 2021-03-15:
@mhummels ESP-AT has supported ESP32-C3 now, please `git pull` the latest code and test

msam-rak, 2021-03-26:
> @mhummels ESP-AT has supported ESP32-C3 now, please `git pull` the latest code and test

@ustccw I tested on ESP32-C3-DevKitM-1 master branch. No reply from UART1. I compiled the same branch on Saola and it is working

mtuxpe, 2021-03-28:
For ESP32-C3 increase stack size

ustccw, 2021-06-21:
@mtuxpe @msam-rak @mhummels ESP32-C3 AT released: https://github.com/espressif/esp-at/releases/tag/v2.2.0.0_esp32c3

this is the first ESP32C3-AT version, without BLE, please know.

ustccw, 2021-06-21:
same to #536

closed 2023-06-05 by ustccw