espressif/esp-at #519: AT+BLUFI example and documentation

rudib, 2021-03-05:
I'm looking for some documentation on how to use the new `AT+BLUFI` command in the 2.1.0.0 release. I've tried various permutations of disabling/enabling WIFI/Bluetooth radios and services, but haven't yet figured it out. Does it need a custom BT server binary file or should it work with the factory binary? A fully worked example would be great.

```
AT+RST
OK
ready
AT+CWMODE=0
OK
AT+BLEINIT=2
OK
AT+BLUFI=1
ERROR
˙˙˙˙

ustccw, 2021-03-12:
@xiewenxiang please take a look

Luxin812, 2021-09-14:
@rudib The ESP-AT firmware used to not enable the BLUFI command by default. At present, the latest ESP-AT master firmware already supports the BLUFI command by default. You can use the latest ESP-AT master to test and develop.

closed 2023-06-05 by ustccw