espressif/esp-at #520: SPI AT for ESP32S2

lgl88911, 2021-03-09:
I pull master at version cf5c761b7b7783c,  can't find ESP AT config option for ESP32.
At the time of 1.19, the configuration items of ESP32S2 SPI AT can be found on the master. Although it was not compiled at that time, please refer to https://github.com/espressif/esp-at/issues/497#issuecomment-762631106. May I ask about ESP32 S2 Will SPI AT be released or has been abandoned

ustccw, 2021-03-12:
@lgl88911 Do you consider using ESP32C3 chip?  esp-at will support it soon.

lgl88911, 2021-03-13:
> @lgl88911 Do you consider using ESP32C3 chip? esp-at will support it soon.

No, I want to consider ESP32S2 SPI AT.

gregrasche, 2022-01-06:
I too want esp32s2 support of esp-at project.  This ticket was 9 months ago. 

This seems to suggest that the esp32s2 is supported
https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp32c3/Compile_and_Develop/How_to_clone_project_and_compile_it.html     

Is esp-at supported on SPI today in esp-idf-v4.3.2?

gregrasche, 2022-01-07:
We have chosen the esp32s2 MINI because it has more I/O and a camera DVP interface.  We cannot use the c3 MINI

If we move to ESP-AT update release/v2.2.0.0_esp32 as suggested at https://github.com/espressif/esp-at  
1) where in GitHub is a SPI driver that we can use?  
2) which esp-idf is it compatible with?
3) are there DOCs on this version of the SPI driver