espressif/esp-at #521: ESP-AT Master AT+CIPCLOSE hangup

lgl88911, 2021-03-10:
I build ESP-AT Master for ESP32-WROVER-E, I can use the AT command to scan, conect, send, recv normally, but after executing AT+CIPCLOSE, esp32 does not respond, and the debug serial port will return some backtrace：

AT+CWJAP="AP8","PASSWORD"  -> Pass
AT+CWMODE=1  -> Pass
AT+CIPMUX=1  -> Pass
AT+CIPSTART=1,"TCP","192.168.101.102",8880  -> Pass
AT+CIPSEND=1,10  -> Pass
AT+CIPCLOSE=1  -> Fail
 show log:
 > Backtrace:0x400D52C9:0x3FFBE3A0 0x40083BEA:0x3FFBE3C0 0x401F5264:0x3FFEA500 0x400E7445:0x3FFEA520 0x40092A16:0x3FFEA5A0

ustccw, 2021-03-12:
could you please provide the `AT+GMR` and detailed log.

H3adcra5h, 2021-03-12:
Same here CIPCLOSE ends up with a reboot after some seconds.
The same behavior if the server closed the connection. There comes no CLOSED, but a reboot after watchdog triggered.

```
Task watchdog got triggered.
Guru Meditation Error: Core  0 panic\'ed (Guru Meditation Error: Core  0 panic\'ed (LoadProhibited). Exception was unhandled

Core 0 register dump:
PC      : 0x402541cb  PS      : 0x00000033  A0      : 0x402543a9  A1      : 0x3ffe81e0  
A2      : 0x00208000  A3      : 0x00000000  A4      : 0x00000018  A5      : 0x00000016 
A6      : 0x402c8572  A7      : 0x00000014  A8      : 0x3ffecb3c  A9      : 0x000000a0  
A10     : 0x40107c4c  A11     : 0x40107c30  A12     : 0x00208000  A13     : 0x402c8571  
A14     : 0x3ffe8200  A15     : 0x3fff6e10  SAR     : 0x00000018  EXCCAUSE: 0x0000001c 

Backtrace: 0x402541cb:0x3ffe81e0 0x402543a9:0x3ffe8200 0x40100f00:0x3ffe8230 0x40225ce4:0x3ffe8270 0x402221bf:0x3ffe82c0 0x40100a12:0x3ffe82d0 0x40100d20:0x3ffe82f0 0x40227b0c:0x3ffe8340 0x402543a9:0x3ffe83b0 
ets Jan  8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load
```

My current version, self compiled, but not modified.

```
AT+GMR
AT version:2.2.0.0-dev(s-dd5ecec - ESP8266 - Feb 24 2021 02:56:03)
SDK version:v3.4-rc
compile time(cf5c761):Mar 11 2021 10:11:38
Bin version:2.0.0(ESP8266_1MB)
```

lgl88911, 2021-03-13:
> could you please provide the `AT+GMR` and detailed log.

AT version:2.2.0.0-dev(s -dd5ecec - ESP32  - Feb 24 2021 0 2:57:29)
SDK version:v4.2
compile  time(cf5c761):Ma r 10 2021 23:52: 38
version: 2.1.0(WROVER-32)

H3adcra5h, 2021-03-13:
I downgraded to this version:
git branch **release/v2.1.0.0_esp8266**
```
AT version:2.1.1.0-dev(24b71e4 - Aug 25 2020 08:41:31)
SDK version:v3.3-2-gb6e861a7
compile time(3355cd8):Mar 13 2021 13:53:57
Bin version:2.1.0(Unknown)
```

and it works.

ustccw, 2021-03-18:
@H3adcra5h `s -dd5ecec` means this firmware compiles with silence mode.

However, silence mode has some known issues to be fixed, please compile with no silence mode.

lgl88911, 2021-03-18:
@ustccw 
Yes disable silence mode, AT++CIPCLOSE work well， Thanks for your support.
···
AT version:2.2.0.0-dev(dd5ecec - ESP32 - Feb 24 2021 02: 56:39)
SDK vers ion:v4.2 compile  time(cf5c761):Mar 18 2021 21:47: 45
Bin version: 2.1.0(WROVER-32)
···

Log
```

I (13) boot: ESP-IDF v4.2 2nd stage bootloader
I (13) boot: compile time 21:42:47
I (13) boot: chip revision: 3
I (15) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (22) boot.esp32: SPI Speed      : 40MHz
I (27) boot.esp32: SPI Mode       : DIO
I (31) boot.esp32: SPI Flash Size : 4MB
I (36) boot: Enabling RNG early entropy source...
I (41) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (59) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (67) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (74) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (82) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (89) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (97) boot: End of partition table
I (101) boot_comm: chip revision: 3, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x2b504 (177412) map
I (191) esp_image: segment 1: paddr=0x0012b52c vaddr=0x3ffbdb60 size=0x03bb4 ( 15284) load
I (198) esp_image: segment 2: paddr=0x0012f0e8 vaddr=0x40080000 size=0x00404 (  1028) load
I (199) esp_image: segment 3: paddr=0x0012f4f4 vaddr=0x40080404 size=0x00b24 (  2852) load
I (208) esp_image: segment 4: paddr=0x00130020 vaddr=0x400d0020 size=0x1368d8 (1272024) map
I (746) esp_image: segment 5: paddr=0x00266900 vaddr=0x40080f28 size=0x19584 (103812) load
I (797) esp_image: segment 6: paddr=0x0027fe8c vaddr=0x400c0000 size=0x00068 (   104) load
I (811) boot: Loaded app from partition at offset 0x100000
I (811) boot: Disabling RNG early entropy source...
max tx power=78,ret=0
2.1.0
delete

```

closed 2021-03-18 by lgl88911