espressif/esp-at #523: AT+CIPDNS not working as expected

H3adcra5h, 2021-03-12:
Documentation says 222.222.67.208 is the default DNS server. But query get +CIPDNS:1,"208.67.222.222".
I think documentation is wrong.

Automatic DNS settings from DHCP isn't stored if SYSSTORE=1. After reset CIPDNS isn't changed. So device always has the wrong DNS server if it is connected via DHCP. If there is no internet connection CIPDOMAIN should not work, but not verified yet.

```
AT+CIPDNS?
+CIPDNS:1,"208.67.222.222"

AT+SYSSTORE=1
AT+CIPDNS=0
AT+RST
AT+CIPDNS?
+CIPDNS:1,"208.67.222.222"
```

My firmware version:

```
AT+GMR
AT version:2.2.0.0-dev(s-dd5ecec - ESP8266 - Feb 24 2021 02:56:03)
SDK version:v3.4-rc
compile time(cf5c761):Mar 11 2021 10:11:38
Bin version:2.0.0(ESP8266_1MB)
```

H3adcra5h, 2021-03-13:
I downgraded to this version:
git branch **release/v2.1.0.0_esp8266**
```
AT version:2.1.1.0-dev(24b71e4 - Aug 25 2020 08:41:31)
SDK version:v3.3-2-gb6e861a7
compile time(3355cd8):Mar 13 2021 13:53:57
Bin version:2.1.0(Unknown)
```

and it works as expected

ustccw, 2021-03-16:
`s-dd5ecec` means the firmware is generated by silence mode, while there is a known issue about it.

could you please test with no silence mode.

closed 2021-06-21 by FayeY