espressif/esp-at #525: AT commands on ESP32-S2-KALUGA1

edbek, 2021-03-25:
How can I enable AT commands on ESP32-S2-KALUGA1 debug via UART1 port of ESP32-S2 ? 
I am trying to install :

CONFIG_AT_UART_PORT_TX_PIN_DEFAULT 43
CONFIG_AT_UART_PORT_RX_PIN_DEFAULT 44

so that it would be possible to transmit and receive AT commands through the USB , but nothing happens.
Tell me, has anyone tried to do this?

xcguang, 2021-03-26:
Do you want to use USB to transmit and receive AT commands? This is in our schedule, but it may take about 2 weeks.

edbek, 2021-03-26:
Yes, that's exactly what I want. Then I wait. Thanks you.

But still I would like to understand why with a simple change of pins for UART1 I can't do anything? Moreover, when debugging, I see that the **esp_at_port_recv_data_notify**() function in the "at_uart_task.c" file is called, but the call to the **at_port_read_data**() function does not happen!