espressif/esp-at #528: InstrFetchProhibited error while use

wu5601309, 2021-04-05:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    ESP-AT: v2.1.0.0_esp32s2-292-g8c0aab8
   ESP-IDF: v4.2-351-gce2c615bb
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

- Operating System:     [Linux]
- Power Supply:         [USB|Battery]

## Problem Description

//Detailed problem description goes here.
InstrFetchProhibited ERROR Cycle and re-cycle

detial log:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7140
load:0x40078000,len:13200
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2-351-gce2c615bb 2nd stage bootloader
I (29) boot: compile time 02:10:25
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed : 40MHz
I (45) boot.esp32: SPI Mode : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label Usage Type ST Offset Length
I (70) boot: 0 phy_init RF data 01 01 0000f000 00001000
I (78) boot: 1 otadata OTA data 01 00 00010000 00002000
I (85) boot: 2 nvs WiFi data 01 02 00012000 0000e000
I (92) boot: 3 at_customize unknown 40 00 00020000 000e0000
I (100) boot: 4 ota_0 OTA app 00 10 00100000 00180000
I (107) boot: 5 ota_1 OTA app 00 11 00280000 00180000
I (115) boot: End of partition table
I (119) boot_comm: chip revision: 1, min. application chip revision: 0
I (126) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x2f514 (193812) map
I (209) esp_image: segment 1: paddr=0x0012f53c vaddr=0x3ffbdb60 size=0x00adc ( 2780) load
I (211) esp_image: segment 2: paddr=0x00130020 vaddr=0x400d0020 size=0x10ec38 (1109048) map
0x400d0020: _stext at ??:?

I (638) esp_image: segment 3: paddr=0x0023ec60 vaddr=0x3ffbe63c size=0x02f04 ( 12036) load
I (644) esp_image: segment 4: paddr=0x00241b6c vaddr=0x40080000 size=0x00404 ( 1028) load
0x40080000: _WindowOverflow4 at /home/dream/esp/esp-at/build/../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (646) esp_image: segment 5: paddr=0x00241f78 vaddr=0x40080404 size=0x1e388 (123784) load
I (708) esp_image: segment 6: paddr=0x00260308 vaddr=0x400c0000 size=0x00064 ( 100) load
I (725) boot: Loaded app from partition at offset 0x100000
I (725) boot: Disabling RNG early entropy source...
module_name:WROOM-32
max tx power=78,ret=0
2.1.0
Guru Meditation Error: Core 0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core 0 register dump:
PC : 0x000d51c4 PS : 0x00060a30 A0 : 0x800908c4 A1 : 0x3ff81c60
A2 : 0x00000000 A3 : 0x00000000 A4 : 0x00000001 A5 : 0x3ffb6ea0
A6 : 0x00000001 A7 : 0x00020d21 A8 : 0x800d5268 A9 : 0x3ffc2170
A10 : 0x000d51c4 A11 : 0x00060a23 A12 : 0x00060a20 A13 : 0x000000a5
A14 : 0x000000a5 A15 : 0x00060023 SAR : 0x0000001d EXCCAUSE: 0x00000014
EXCVADDR: 0x000d51c4 LBEG : 0x00000000 LEND : 0x00000000 LCOUNT : 0x00000000

Backtrace:0x000d51c1:0x3ff81c60 |<-CORRUPTED

ELF file SHA256: 323beda5d4225a52

Rebooting...
ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7140
load:0x40078000,len:13200
load:0x40080400,len:4568
0x40080400: _init at ??:?

ustccw, 2021-04-07:
@wu5601309 It seems that the PC register has been corrupted. Could you please `git pull` latest code and test.

If the issue still exists, please provide detailed compilation log and AT commands you input.

wu5601309, 2021-04-08:
@ustccw thank for you reply, and I try git pull latest code, but still have this problem.
I test two way, one way is Enable silence mode to remove some logs and reduce the firmware size. other is disable.
**1) disable silence mode  log**
entry 0x400806f4
I (29) boot: ESP-IDF v4.2-351-gce2c615bb 2nd stage bootloader
I (29) boot: compile time 12:17:33
I (29) boot: chip revision: 1
I (33) boot.esp32: SPI Speed      : 40MHz
I (38) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (78) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (85) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (93) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (100) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x2c800 (182272) map
I (191) esp_image: segment 1: paddr=0x0012c828 vaddr=0x3ffbdb60 size=0x037f0 ( 14320) load
I (197) esp_image: segment 2: paddr=0x00130020 vaddr=0x400d0020 size=0xf3f94 (999316) map
0x400d0020: _stext at ??:?

I (578) esp_image: segment 3: paddr=0x00223fbc vaddr=0x3ffc1350 size=0x00160 (   352) load
I (578) esp_image: segment 4: paddr=0x00224124 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/dream/esp/esp-at/build/../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (585) esp_image: segment 5: paddr=0x00224530 vaddr=0x40080404 size=0x1cdf4 (118260) load
I (645) esp_image: segment 6: paddr=0x0024132c vaddr=0x400c0000 size=0x00064 (   100) load
I (661) boot: Loaded app from partition at offset 0x100000
I (661) boot: Disabling RNG early entropy source...
module_name:WROOM-32
max tx power=78,ret=0
2.1.0
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception).
Debug exception reason: BREAK instr
Core  0 register dump:
PC      : 0x400803c0  PS      : 0x00060c36  A0      : 0x0008f860  A1      : 0x3ff81ba0
0x400803c0: _DoubleExceptionVector at /home/dream/esp/esp-at/build/../esp-idf/components/freertos/xtensa/xtensa_vectors.S:455

A2      : 0x00060c30  A3      : 0x00040023  A4      : 0x00060c20  A5      : 0x08576c70
A6      : 0x08f7cd10  A7      : 0x00040000  A8      : 0x00000001  A9      : 0x0008f860
A10     : 0x00000003  A11     : 0x00060c23  A12     : 0x400829a5  A13     : 0x00040000
0x400829a5: _xt_lowint1 at /home/dream/esp/esp-at/build/../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1098

A14     : 0x00060c20  A15     : 0x00040000  SAR     : 0x0000001d  EXCCAUSE: 0x00000001
EXCVADDR: 0x0008f860  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x400803bd:0x3ff81ba0 |<-CORRUPTED
0x400803bd: _UserExceptionVector at ??:?

ELF file SHA256: d56f6bd7d8b65852

Rebooting...
ets Jun  8 2016 00:22:57
**2)enable silence mode** 
entry 0x400806f4
I (29) boot: ESP-IDF v4.2-351-gce2c615bb 2nd stage bootloader
I (29) boot: compile time 12:28:00
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 phy_init         RF data          01 01 0000f000 00001000
[... 33 lines cut ...]

0x400e6cf5: at_is_transmission_send_mode at ??:?

0x4008f662: vPortTaskWrapper at /home/dream/esp/esp-at/build/../esp-idf/components/freertos/xtensa/port.c:143

Backtrace:0x400D459D:0x3FFBE370 0x400827AE:0x3FFBE390 0x401CE974:0x3FFD86C0 0x400E6CF5:0x3FFD86E0 0x4008F662:0x3FFD8760
0x400d459d: task_wdt_isr at /home/dream/esp/esp-at/build/../esp-idf/components/esp_common/src/task_wdt.c:187

0x400827ae: _xt_lowint1 at /home/dream/esp/esp-at/build/../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x401ce974: at_sys_load_mode_is_flash at ??:?

0x400e6cf5: at_is_transmission_send_mode at ??:?

0x4008f662: vPortTaskWrapper at /home/dream/esp/esp-at/build/../esp-idf/components/freertos/xtensa/port.c:143

use AT cmd:
AT+CWMODE=3
AT+GMR
AT+CIPMUX=1
AT+WEBSERVER=1,80,50 
AT+CIPDOMAIN="cn.ntp.org.cn"
AT+CIPDOMAIN="ntp.rt-thread.org"
AT+CIPDOMAIN="edu.ntp.org.cn"
AT+CIPCLOSE=0

wu5601309, 2021-04-08:
**compilation log:**
dream@dream:~/esp/esp-at$ ./build.py menuconfig
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP8266
3. PLATFORM_ESP32S2
4. PLATFORM_ESP32C3
choose(range[1,4]):1

Module name:
1. WROOM-32
2. WROVER-32
3. PICO-D4
4. SOLO-1
5. MINI-1 (description: ESP32-U4WDH chip inside)
6. ESP32-D2WD (description: 2MB flash, No OTA)
choose(range[1,6]):1

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):1
platform_name=ESP32,module_name=WROOM-32
Executing action: menuconfig
Running cmake in directory /home/dream/esp/esp-at/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/dream/esp/esp-at"...
silence:1 sdkconfig:/home/dream/esp/esp-at/module_config/module_esp32_default/sdkconfig_silence.defaults
-- Found Git: /usr/bin/git (found version "2.27.0")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/dream/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/dream/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/dream/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/dream/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/dream/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/dream/esp/esp-at/sdkconfig
Loading defaults file /home/dream/esp/esp-at/module_config/module_esp32_default/sdkconfig_silence.defaults...
-- Found PythonInterp: /home/dream/.espressif/python_env/idf4.2_py3.8_env/bin/python (found version "3.8.6")
-- Found Perl: /usr/bin/perl (found version "5.30.3")
-- App "esp-at" version: v2.1.0.0_esp32s2-318-g11b35ee-d
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-nano.ld
-- Adding linker script /home/dream/esp/esp-at/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/dream/esp/esp-at/esp-idf/components/esp32/ld/esp32.peripherals.ld
silence:1, ld core:/home/dream/esp/esp-at/components/at/lib/libesp32_at_core_silence.a
Parsing CSV input...
Verifying table...
-- Components: app_trace app_update asio at bootloader bootloader_support bt cbor coap console customized_partitions cxx decoder driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_driver spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/dream/esp/esp-at/esp-idf/components/app_trace /home/dream/esp/esp-at/esp-idf/components/app_update /home/dream/esp/esp-at/esp-idf/components/asio /home/dream/esp/esp-at/components/at /home/dream/esp/esp-at/esp-idf/components/bootloader /home/dream/esp/esp-at/esp-idf/components/bootloader_support /home/dream/esp/esp-at/esp-idf/components/bt /home/dream/esp/esp-at/esp-idf/components/cbor /home/dream/esp/esp-at/esp-idf/components/coap /home/dream/esp/esp-at/esp-idf/components/console /home/dream/esp/esp-at/components/customized_partitions /home/dream/esp/esp-at/esp-idf/components/cxx /home/dream/esp/esp-at/components/decoder /home/dream/esp/esp-at/esp-idf/components/driver /home/dream/esp/esp-at/esp-idf/components/efuse /home/dream/esp/esp-at/esp-idf/components/esp-tls /home/dream/esp/esp-at/esp-idf/components/esp32 /home/dream/esp/esp-at/esp-idf/components/esp_adc_cal /home/dream/esp/esp-at/esp-idf/components/esp_common /home/dream/esp/esp-at/esp-idf/components/esp_eth /home/dream/esp/esp-at/esp-idf/components/esp_event /home/dream/esp/esp-at/esp-idf/components/esp_gdbstub /home/dream/esp/esp-at/esp-idf/components/esp_hid [... cut ...] ons/server_key.bin 0x28000 build/customized_partitions/server_ca.bin 0x2a000 build/customized_partitions/client_cert.bin 0x2c000 build/customized_partitions/client_key.bin 0x2e000 build/customized_partitions/client_ca.bin 0x30000 build/customized_partitions/factory_param.bin 0x37000 build/customized_partitions/mqtt_cert.bin 0x39000 build/customized_partitions/mqtt_key.bin 0x3B000 build/customized_partitions/mqtt_ca.bin 0x100000 build/esp-at.bin
or run 'idf.py -p (PORT) flash'
idf.py build ret: 0

ustccw, 2021-04-28:
@wu5601309 could you please download the esp-at firmware from GitHub Actions: https://github.com/espressif/esp-at/actions/runs/789286795#artifacts

To check whether your hardware is ready or not.

Moreover, please do not use silence mode to compile, for it is a ongoing work.

FayeY, 2021-06-21:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2021-06-21 by FayeY