espressif/esp-at #529: 启用ESP-AT Ethernet后，重新编译的固件运行报错，且不断重启。我该如何解决？

wajlh, 2021-04-05:
如题所述：模组为ESP32-WROOM-32UE
sdkconfig如附件：[sdkconfig.zip](https://github.com/espressif/esp-at/files/6259692/sdkconfig.zip)

参照如下链接设置：https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_enable_ESP_AT_Ethernet.html

串口日志如下：

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7140
load:0x40078000,len:13200
load:0x40080400,len:4568
entry 0x400806f4
[0;32mI (29) boot: ESP-IDF v4.2-351-gce2c615bb 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 00:15:00[0m
[0;32mI (29) boot: chip revision: 3[0m
[0;32mI (33) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (38) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (42) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Enabling RNG early entropy source...[0m
[0;32mI (52) boot: Partition Table:[0m
[0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (63) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (71) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (78) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (86) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (93) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (100) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (108) boot: End of partition table[0m
[0;32mI (112) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x21f88 (139144) map[0m
[0;32mI (174) esp_image: segment 1: paddr=0x00121fb0 vaddr=0x3ffb0000 size=0x028b4 ( 10420) load[0m
[0;32mI (179) esp_image: segment 2: paddr=0x0012486c vaddr=0x40080000 size=0x00404 (  1028) load[0m
[0;32mI (181) esp_image: segment 3: paddr=0x00124c78 vaddr=0x40080404 size=0x0b3a0 ( 45984) load[0m
[0;32mI (210) esp_image: segment 4: paddr=0x00130020 vaddr=0x400d0020 size=0xb75c8 (751048) map[0m
[0;32mI (496) esp_image: segment 5: paddr=0x001e75f0 vaddr=0x4008b7a4 size=0x0af4c ( 44876) load[0m
[0;32mI (516) esp_image: segment 6: paddr=0x001f2544 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (529) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (529) boot: Disabling RNG early entropy source...[0m
max tx power=78,ret=0
2.1.0
[0;31mE (683) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffaf350 (config or if_key is NULL or duplicate key)[0m
[0;31mE (684) esp_eth.netif.glue: esp-netif handle can't be null[0m
ESP_ERROR_CHECK failed: esp_err_t 0x102 at 0x40086db8
file: "../components/at/src/at_eth_init.c" line 32
func: esp_at_get_eth_default_config
expression: tcpip_adapter_set_default_eth_handlers()

abort() was called at PC 0x40086dbb on core 0

Backtrace:0x40087efb:0x3ffaf2b0 0x40088519:0x3ffaf2d0 0x4008f29a:0x3ffaf2f0 0x40086dbb:0x3ffaf360 0x400e7e94:0x3ffaf380 0x400d96b1:0x3ffaf3e0 0x400d619d:0x3ffaf420 0x400d423f:0x3ffaf440 0x40088521:0x3ffaf470

ELF file SHA256: 7696f1d15c478c7f

Rebooting...

ustccw, 2021-04-28:
@ESP-Coco could you please to take a look.

ustccw, 2021-06-21:
@wajlh sorry for late reply. esp-at ethernet issue got fixed on latest version. please test.

closed 2021-07-30 by FayeY