espressif/esp-at #531: MQTT Publish Crash on ESP8266

leungngok, 2021-04-09:
**Environment**
chip: ESP8266
version: v2.1.0.0_esp8266
connect with STM32 via UART

**Problem**
If ESP8266 receives two messages via MQTT subscription very very closely and STM32 responses with AT+MQTTCONN immediately, ESP8266 will reply "busy p...", and keep stuck until I unplug CHIP_EN to reboot. After reboot, it will show "ready".

MQTT Config: QoS1, auto reconnect (maybe using more ram?).

It is caused by depletion of ram? How to jump out of "busy" except for reboot, and what's the best way to reboot?

ustccw, 2021-04-27:
@leungngok please use the latest version. if the issue still exists, provide the `AT+GMR` for us.

ustccw, 2021-04-27:
you can download the latest firmware at https://github.com/espressif/esp-at/actions/runs/782418171#artifacts

closed 2021-06-21 by FayeY