espressif/esp-at #533: Failed to verify partition table

leungngok, 2021-04-16:
## Environment

- Chip:  ESP8266
- esp-at: master
- IDF version : 3.4
- Build System: both Windows and WSL2 (Ubuntu)
- Compiler version: xtensa-lx106-elf-gcc8_4_0-esp-2020r3-win32 / xtensa-lx106-elf-gcc8_4_0-esp-2020r3-linux-amd64

## Problem Description
Unable to boot.

Menuconfig:
Default custom partition table and default at customized partition table.
Console output UART0.
Same log output using DIO and QIO SPI mode.

##LOG
 boot: ESP-IDF v3.4 2nd stage bootloader
[0;32mI (51) boot: compile time 15:31:28
[0;32mI (51) boot: SPI Speed      : 80MHz
[0;32mI (54) boot: SPI Mode       : DIO
[0;32mI (58) boot: SPI Flash Size : 2MB
[0;31mE (62) flash_parts: partition 0 invalid magic number 0xf1f1
[0;31mE (68) boot: Failed to verify partition table
[0;31mE (73) boot: load partition table error!
user code done

ustccw, 2021-04-19:
@leungngok it seems that esp-at used the wrong esp-idf version from your log. It should be `v3.4-rc`, other than `v3.4`.

could you please test the startup without any modifications.  
It the issue still exists, please provide the detailed compile log and startup log.

Thanks.
CW

leungngok, 2021-04-22:
Updated idf v3.4 commit version, but issue still exists.
Default menuconfig setting except that console log output is set to UART0 to print startup log.
The UART0 Tx/Rx pin is set to GPIO1/3 in factory_param_data.csv

**Compile log**

PS C:\Users\JHL\Desktop\Workspace\esp-at> python ./build.py build
module_name JHL
platform_name=ESP8266,module_name=JHL
Setting IDF_PATH environment variable: c:\users\jhl\desktop\workspace\esp-at\esp-idf
Checking Python dependencies...
Python requirements from c:\users\jhl\desktop\workspace\esp-at\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\users\jhl\desktop\workspace\esp-at\build
Executing "ninja all"...
[11/614] Generating ../../phy_init_data.bin
C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/esp8266/include/esp_private/phy_init_data.h:15:9: warning: #pragma once in main file
 #pragma once
         ^~~~
[122/614] Performing configure step for 'bootloader'
CMake Warning at C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/tools/cmake/idf.cmake:14 (message):
  IDF_PATH environment variable is different from inferred IDF_PATH.

                              Check if your project's top-level CMakeLists.txt includes the right
                              CMake files. Environment IDF_PATH will be used for the build.
Call Stack (most recent call first):
  C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:20 (include)

-- Project version: v3.4-4-ge2fe1803
-- Building ESP-IDF components for target esp8266
-- Adding linker script C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/bootloader/subproject/main/esp8266.bootloader.ld
-- Adding linker script C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/bootloader/subproject/main/esp8266.bootloader.rom.ld
-- Adding linker script C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/esp8266/ld/esp8266.rom.ld
-- Components: bootloader bootloader_support esp8266 esp_common esptool_py log main partition_table spi_flash
-- Component paths: C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/bootloader C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/bootloader_support C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/esp8266 C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/esp_common C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/esptool_py C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/log C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/bootloader/subproject/main C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/partition_table C:/Users/JHL/Desktop/Workspace/esp-at/esp-idf/components/spi_flash
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/JHL/Desktop/Workspace/esp-at/build/bootloader
[265/614] Performing build step for 'bootloader'
[1/2] Linking C executable bootloader.elf
[2/2] Generating binary image from built executable
esptool.py v2.4.0
Generated C:/Users/JHL/Desktop/Workspace/esp-at/build/bootloader/bootloader.bin
[612/614] Generating binary image from built executable
esptool.py v2.4.0
Generated C:/Users/JHL/Desktop/Workspace/esp-at/build/esp-at.bin
[613/614] Running utility command for customized_bin
generating server_cert.bin: C:\Users\JHL\anaconda3\python.exe C:/Users/JHL/Desktop/Workspace/esp-at\tools\AtPKI.py generate_bin -b C:/Users/JHL/Desktop/Workspace/esp-at/build/customized_partitions\server_cert.bin  cert C:/Users/JHL/Desktop/Workspace/esp-at\components\customized_partitions\raw_data\server_cert\server_cert.crt
generating server_key.bin: C:\Users\JHL\anaconda3\python.exe C:/Users/JHL/Desktop/Workspace/esp-at\tools\AtPKI.py generate_bin -b C:/Users/JHL/Desktop/Workspace/esp-at/build/customized_partitions\server_key.bin  cert C:/Users/JHL/Desktop/Workspace/esp-at\components\customized_partitions\raw_data\server_key\server.key
generating server_ca.bin: C:\Users\JHL\anaconda3\python.exe C:/Users/JHL/Desktop/Workspace/esp-at\tools\AtPKI.py generate_bin -b C:/Users/JHL/Desktop/Workspace/esp-at/build/customized_partitions\server_ca.bin  cert C:/Users/JHL/Desktop/Workspace/esp-at\components\customized_partitions\raw_data\server_ca\server_ca.crt
generating client_cert.bin: C:\Users\JHL\anaconda3\python.exe C:/Users/JHL/Desktop/Workspace/esp-at\tools\AtPKI.py generate_bin -b C:/Users/JHL/Desktop/Workspace/esp-at/build/customized_partitions\client_cert.bin  cert C:/Users/JHL/Desktop/Workspace/esp-at\components\customized_partitions\raw_data\client_cert\client_cert_00.crt cert C:/Users/JHL/Desktop/Workspace/esp-at\components\customized_partitions\raw_data\client_cert\client_cert_01.crt
generating client_key.bin: C:\Users\JHL\anaconda3\python.exe C:/Users/JHL/Desktop/Workspace/esp-at\tools\AtPKI.py generate_bin -b C:/Users/JHL/Desktop/Workspace/esp-at/build/customized_partitions\client_key.bin  cert C:/Users/JHL/Desktop/Workspace/esp-at\components\customized_partitions\raw_data\client_key\client_key_00.key cert C:/Users/JHL/Desktop/Workspace/esp-at\components\customized_partitions\raw_data\client_key\client_key_01.key
generating client_ca.bin: C:\Users\JHL\anaconda3\python.exe C:/Users/JHL/Desktop/Workspace/esp-at\tools\AtPKI.py generate_bin -b C:/Users/JHL/Desktop/Workspace/esp-at/build/customized_partitions\client_ca.bin  cert C:/Users/JHL/Desktop/Workspace/esp-at\components\customized_partitions\raw_data\client_ca\client_ca_00.crt cert C:/Users/JHL/Desktop/Workspace/esp-at\components\c [... cut ...] 000 build\customized_partitions\mqtt_ca.bin
or run 'idf.py -p (PORT) flash'
idf.py build ret: 0

**Startup log**
boot: ESP-IDF v3.4-4-ge2fe1803 2nd stage bootloader
0;32mI (52) boot: compile time 11:53:50
0;32mI (52) boot: SPI Speed      : 80MHz
0;32mI (55) boot: SPI Mode       : DIO
0;32mI (59) boot: SPI Flash Size : 2MB
0;31mE (63) flash_parts: partition 0 invalid magic number 0xf1f1
0;31mE (69) boot: Failed to verify partition table
0;31mE (74) boot: load partition table error!
user code done

ustccw, 2021-04-27:
@leungngok It seems that you are using your own module:`JHL` configuration.

Could you please follow the guide: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html#esp8266-platform

Test whether the native ESP8266 configuration works or not in your env. This ensures that your hardware and software are ready to work before using your custom configuration.

FayeY, 2021-06-21:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2021-06-21 by FayeY