espressif/esp-at #535: “AT+SAVETRANSLINK ” can not save IP 255.255.255.255

xuhongv, 2021-04-27:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- AT version : 

```
AT version:2.2.0.0-dev(30100ff - ESP32 - Dec  7 2020 10:58:40)
SDK version:v4.0.1-193-ge7ac221
compile time(abc25a8):Dec  8 2020 15:09:36
Bin version:2.1.0(WROOM-32)
```

## Problem Description

ESP device will enter the Wi-Fi passthrough mode .

```
AT+CIPSTART="UDP","255.255.255.255",6123,8989,0
CONNECT

OK
AT+CIPMODE=1

OK
AT+CIPSEND

OK

>
```

Now I  want the  ESP device will enter the Wi-Fi passthrough mode in any subsequent power cycles  ,  so：

```
AT+SAVETRANSLINK=1,"255.255.255.255",1002,"UDP"

ERROR
```

Why not the "255.255.255.255" is error , but in the AT+CIPSTART is OK ?

ustccw, 2021-04-27:
@xuhongv it seems that you're using an older version, and the new version supports this feature.

xuhongv, 2021-04-27:
HI 我刚刚更新了一版 ：

```
AT+GMR
AT version:2.2.0.0-dev(s-3ac9d0f - ESP8266 - Apr 19 2021 08:05:35)
SDK version:v3.4
Ai-Thinker-compile time(233da6c):Apr 27 2021 01:56:20
Bin version:2.0.0(WROOM-02-N)

OK

```

我这样的操作，比上个版本更糟糕了，而且发现无法透传，也无法退出头窜，然后不断重启。

```
AT+CWMODE=1

OK
AT+CWJAP="aiot@xuhongv","xuhong12345678"
WIFI CONNECTED
WIFI GOT IP

OK
AT+SAVETRANSLINK=1,"255.255.255.255",6123,"UDP"

OK
{dl湠?宑鉶?€l?b|巸靝搒揷?"勻NN焏NN溿?Bp寧cl`slp鹢??lbn鈢?巪?"勻NN?$剰$ ?ool`or?簿s{屲lp鹢?s屲茆?"N骡N?廈勻NN?l`?2ool`o{彌沷冦l`p髇?s屲溌?"N骡N??勻NN?d ?ool`o{彌沷c屲l{洓oc屲l
>
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 6960, room 16 
tail 0
chksum 0x3f
load 0x3ffe8008, len 24, room 8 
tail 0
chksum 0xd6
load 0x3ffe8020, len 3184, room 8 
tail 8
chksum 0x2f
csum 0x2f

>
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 6960, room 16 
tail 0
chksum 0x3f
load 0x3ffe8008, len 24, room 8 
tail 0
chksum 0xd6
load 0x3ffe8020, len 3184, room 8 
tail 8
chksum 0x2f
csum 0x2f

>
```

ustccw, 2021-04-27:
Please do not use silence mode compilation at the beginning of the compile. 
silence mode is ongoing work internally now.

ustccw, 2021-04-27:
It is recommended that download esp-at common firmware from GitHub Actions: https://github.com/espressif/esp-at/actions/runs/782418171#artifacts

ustccw, 2021-06-21:
@ustccw silence mode should be okey now. please get the latest code and test.

closed 2021-07-30 by FayeY