espressif/esp-at #536: ESP32 C3 support please. WiFi is working but not BLE

jamesjian007, 2021-04-28:
I am using ESP32 C3 from ai-thinker and ESP32-C3-DevKitM-1 .

ustccw, 2021-04-28:
@jamesjian007 `ESP32C3-AT` support BLE is an internal ongoing work. 

The first `ESP32C3-AT` firmware, without BLE, is planned to be released in the next month.
It is planned to release the second version of `ESP32C3-AT`t firmware in the next two months for BLE.

> BLE in ESP32C3-AT will use new  AT BLE commands, different from ESP32-AT, for ESP32C3 will use NimBLE.

ustccw, 2021-06-21:
@jamesjian007 ESP32-C3 AT released: https://github.com/espressif/esp-at/releases/tag/v2.2.0.0_esp32c3

this is the first ESP32C3-AT version, without BLE, please know.

leungngok, 2021-07-15:
Any latest news about the second version of ESP32C3-AT? Waiting for BLE support.

ustccw, 2021-07-19:
@leungngok Sorry for possible inconvenience. Here are latest updates from internal.
- we are in testing ESP32-C3 BLE internally now
- ESP32-C3 firmware with BLE will be available in latest month (without full-test internal)
- ESP32-C3 firmware with BLE is planned to be released around the end of october
- ESP32-C3 AT BLE commands should be same as ESP32

please know.

mtuxpe, 2021-07-24:
> @jamesjian007 `ESP32C3-AT` support BLE is an internal ongoing work.
> 
> The first `ESP32C3-AT` firmware, without BLE, is planned to be released in the next month.
> It is planned to release the second version of `ESP32C3-AT` firmware in the next two months for BLE.
> 
> > BLE in ESP32C3-AT will use new  AT BLE commands, different from ESP32-AT, for ESP32C3 will use NimBLE.

mtuxpe, 2021-07-24:
> @jamesjian007 ESP32-C3 AT released: https://github.com/espressif/esp-at/releases/tag/v2.2.0.0_esp32c3
> 
> this is the first ESP32C3-AT version, without BLE, please know.

My target hardware  is ESP32-C3-DevKitM-1 v1.0  and the revision is 2: ESP32C3_REV_MIN_2 on menuconfig.
ESP-IDF version: ESP-IDF v4.3-beta3-195-g6be10fab0

UART AT console:
ready
AT+GMR

(No reply to AT commands)

ustccw, 2021-07-29:
@mtuxpe sorry, `ESP32-C3` ECO2 (the revision is 2) chip is not supported, for it is not the massive product chip.  
please use `ESP32-C3` ECO3 (the revision is 3) chip instead.

donghang6, 2021-11-10:
Is the `ESP32-C3` firmware with BLE now?

ustccw, 2021-11-11:
@donghang6 yes, ESP32-C3 AT supports BLE now, you can [Download the Latest Temporary Version of AT Firmware from GitHub](https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_download_the_latest_temporary_version_of_AT_from_github.html) or [Compile ESP-AT Project](https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html).

btw, we are in first round testing for ESP32-C3 release internally.

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

closed 2022-05-27 by jamesjian007

mtuxpe, 2022-05-27:
Congrats. BLE is working
AT+GMR
AT version:2.4.0.0(s-4c6eb5e - ESP32C3 - May 20 2022 03:12:30)
SDK version:v4.3.2-553-g588436db47
compile time(40ae56a):May 27 2022 11:29:02
Bin version:2.4.0(MINI-1)
Tested on ESP32-C3-DevKitC-02 and ESP32-C3S_Kit