espressif/esp-at #539: AT+MQTTPUBRAW 返回数据格式错误

edieruby, 2021-05-19:
## Environment

- Development Kit:      none
- Module or chip used:  ESP-WROOM-02D
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):  
[image]

- Power Supply:         external 5V

## Problem Description
MQTT 进行批量数据测试, 即收到定义的消息后,立刻pub该消息.
相当于同时收发消息.

测试时连续收到100条消息

### Expected Behavior
ESP8266 RX 日志如下:
[image]

ESP8266 TX 日志如下:
[image]

在最后的AT+MQTTPUBRAW TX数据,
[image]

应当返回
[image]
响应,
但是在OK后插入了一条MQTTSUBRECV的数据, 数据发送格式错误.

### Actual Behavior

正确应该是返回
[image]
中间不会有event插入.

非必现,但是复现概率很高.

ustccw, 2021-05-24:
@edieruby i am afraid ESP-AT can't do that now.
`OK` and `>` are the prompt for AT TX, and `+MQTTSUBRECV:xxx` is the prompt for AT RX. they are separate.

As far as  `+MQTTSUBRECV:<id>,<len>,<topic>,<data>` is concerned, it should not be interrupted by any others.

From your side, i would like to suggest adding more parsing logic to distinguish TX and RX information of AT command port.

edieruby, 2021-05-25:
@ustccw  你好:
ok > 和 +MQTTSUBRECV:xxx 都是在AT TX上输出.

在ok > 输出时, 是不是应该将他看作一个整体进行输出, +MQTTSUBRECV:xxx应该是在整个输出结束后, 或者还没开始前就输出, 
不应该在ok和>的中间进行输出?

测试的时候 目前也只发现 在ok\r\n 后会插入+MQTTSUBRECV:xxx 
按照你的意思 有可能在输出 o 后也有可能插入+MQTTSUBRECV:xxx ? 然后再输出k\r\n? 这种我还没遇到过,
测试的时候 感觉ok和> 是被当成了两个整体, 只会在ok\r\n后会被中断, 在中间不会被中断.

ustccw, 2021-05-25:
@edieruby please test with the following *.a, instead of https://github.com/espressif/esp-at/tree/master/components/at/lib

[libespat.zip](https://github.com/espressif/esp-at/files/6537164/libespat.zip)

If it works, will auto sync to GitHub after internal test cases passed.

ustccw, 2021-05-25:
The original design about the "OK" and ">" is like as the explanation here: https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/index.html#at-messages.

- "OK" means "AT command process done and return OK"
- ">" means "ESP-AT is waiting for more data to be received"

edieruby, 2021-05-25:
好的, 那我就按照分开的,自己代码处理下就好了. 两种都能支持.

closed 2021-05-25 by edieruby