espressif/esp-at #540: AT+CWLAPOPT gives ERR CODE:0x01060204 when using optional parameters

jortbmd, 2021-05-21:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32
- Power Supply:        external 5V
- AT version: version:2.1.0.0(883f7f2 - Jul 24 2020 11:50:07)
- SDK version SDK version:v4.0.1-193-ge7ac221
- Bin version: 2.1.0(WROOM-32) 

## Problem Description
When using the AT+CWLAPOPT command on the default AT firmware gives ERR CODE:0x01060204 when using optional parameters

e.g. sending the message: "AT+CWLAPOPT=1,31" works fine and results in and OK message. 
Sending: "AT+CWLAPOPT=1,31,-100" result in the error message: ERR CODE:0x01060203
Sending: "AT+CWLAPOPT=1,31,,255" results in the error message: ERR CODE:0x01060204
Sending: "AT+CWLAPOPT=1,31,-100,255" result in the error message: ERR CODE:0x01060204

### Expected Behavior
To have OK returned instead of the error messages.

### Actual Behavior
Returns error messages when including optional parameters.
### Steps to repropduce

1. send "AT+CWLAPOPT=1,31,-100" as an AT messagge

ustccw, 2021-05-24:
@jortbmd according to the err code described here, https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/AT_API_Reference.html#_CPPv417esp_at_error_code
0x01060203 means: command parameter counts mismatch (ESP_AT_SUB_PARA_NUM_MISMATCH), that is, expected two parameters, but three parameters were parsed.  
0x01060204 means expected two parameters, but four parameters were parsed.

Actually, the v2.1.0 AT firmware, AT+CWLAPOPT just supports two parameters, see as:
https://docs.espressif.com/projects/esp-at/en/release-v2.1.0.0_esp32/AT_Command_Set/Wi-Fi_AT_Commands.html#cmd-LAPOPT

the latest AT firmware can support to four parameters, see as:
https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/Wi-Fi_AT_Commands.html#cmd-lapopt

Please get the latest firmware from the GitHub Actions: https://github.com/espressif/esp-at/actions or compile AT firmware according to https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html

jortbmd, 2021-05-24:
Thank you! This helps. I could not find any updated binaries for the release. So I assumend it was the latest.

closed 2021-05-24 by jortbmd