espressif/esp-at #541: ESP32 AT not responsed until reset

majianjia, 2021-05-30:
## Environment

~~~
<6>May 30 19:15:56 at.dev.esp32: esp0 device initialize retry...
<6>May 30 19:15:56 at.dev.esp32: esp0 device wifi is disconnect.
<6>May 30 19:15:57 at.dev.esp32: AT version:2.1.0.0(883f7f2 - Jul 24 2020 11:50:07)
<6>May 30 19:15:57 at.dev.esp32: SDK version:v4.0.1-193-ge7ac221
<6>May 30 19:15:57 at.dev.esp32: compile time(0ad6331):Jul 28 2020 02:47:21
<6>May 30 19:15:57 at.dev.esp32: Bin version:2.1.0(WROOM-32)
~~~
Power supply 3.3V.

## Problem Description

After a few hours of pressure MQTT test, it stops responding to any AT command. 

It also fails to send a message which leads to closing and reopening the socket. Every a few minutes (from 5 to 20min).
Until finally it stops responding to any AT command. (also tested by hand writing `AT`)

### Expected Behavior

It should reply error/busy/OK no matter the state of the module.

### Actual Behavior

No response.

### Steps to repropduce

### Code to reproduce this issue

## Debug Logs

```
<3>May 30 18:03:34 at.skt.esp32: esp0 device socket(0) wait connect result timeout.
<7>May 30 18:03:54 at.dev.esp32: system is processing a commands...
<3>May 30 18:04:54 mqtt: [536968768] can't recv Ping Response
<7>May 30 18:04:54 mqtt: MQTT offline
<3>May 30 18:04:54 mqtt: publish fail, wait for reconnect
<7>May 30 18:04:59 mqtt: restart!
<7>May 30 18:04:59 mqtt: MQTT connecting
<7>May 30 18:04:59 mqtt: ipv4 address port: 1883
<7>May 30 18:04:59 mqtt: HOST = '192.168.1.87'
<6>May 30 18:04:59 mqtt: MQTT server connect success.
<6>May 30 18:04:59 mqtt: Subscribe #0 state OK!
<6>May 30 18:04:59 mqtt: Subscribe #1 ota_downstream OK!
<7>May 30 18:04:59 mqtt: MQTT online

msh />
msh /><7>May 30 18:14:01 at.dev.esp32: system is processing a commands...
<3>May 30 18:14:11 at.skt.esp32: esp0 device socket(0) wait connect result timeout.
<3>May 30 18:14:31 mqtt: publish fail, wait for reconnect
<7>May 30 18:14:36 at.clnt: execute command (AT+CIPSEND=0,2) timeout (5000 ticks)!
<3>May 30 18:14:36 mqtt: [11636394] send ping rc: -1 
<7>May 30 18:14:36 at.dev.esp32: system is processing a commands...
<7>May 30 18:14:36 mqtt: MQTT offline
<7>May 30 18:14:41 mqtt: restart!
<7>May 30 18:14:41 mqtt: MQTT connecting
<7>May 30 18:14:41 mqtt: ipv4 address port: 1883
<7>May 30 18:14:41 mqtt: HOST = '192.168.1.87'
<6>May 30 18:14:41 mqtt: MQTT server connect success.
<6>May 30 18:14:42 mqtt: Subscribe #0 state OK!
<6>May 30 18:14:42 mqtt: Subscribe #1 ota_downstream OK!
<7>May 30 18:14:42 mqtt: MQTT online
<7>May 30 18:19:12 at.dev.esp32: system is processing a commands...
<3>May 30 18:19:22 at.skt.esp32: esp0 device socket(0) wait connect result timeout.
<7>May 30 18:19:42 at.dev.esp32: system is processing a commands...
<3>May 30 18:20:42 mqtt: [536968768] can't recv Ping Response
<7>May 30 18:20:42 mqtt: MQTT offline
<3>May 30 18:20:43 mqtt: publish fail, wait for reconnect
<7>May 30 18:20:47 mqtt: restart!
<7>May 30 18:20:47 mqtt: MQTT connecting
<7>May 30 18:20:47 mqtt: ipv4 address port: 1883
<7>May 30 18:20:47 mqtt: HOST = '192.168.1.87'
<6>May 30 18:20:48 mqtt: MQTT server connect success.
<6>May 30 18:20:48 mqtt: Subscribe #0 state OK!
<6>May 30 18:20:48 mqtt: Subscribe #1 ota_downstream OK!
<7>May 30 18:20:48 mqtt: MQTT online
<3>May 30 18:23:44 at.skt.esp32: esp0 device socket(0) wait connect result timeout.
<3>May 30 18:23:48 mqtt: publish fail, wait for reconnect
<7>May 30 18:23:53 at.clnt: execute command (AT+CIPSEND=0,2) timeout (5000 ticks)!
<3>May 30 18:23:53 mqtt: [12193202] send ping rc: -1 
<7>May 30 18:23:58 at.clnt: execute command (AT+CIPSEND=0,2) timeout (5000 ticks)!
<7>May 30 18:23:58 mqtt: MQTT offline
<7>May 30 18:23:58 at.clnt: execute command (AT+CIPCLOSE=0) timeout (300 ticks)!
<7>May 30 18:24:04 mqtt: restart!
<7>May 30 18:24:04 mqtt: MQTT connecting
<7>May 30 18:24:04 mqtt: ipv4 address port: 1883
<7>May 30 18:24:04 mqtt: HOST = '192.168.1.87'
<7>May 30 18:24:09 at.clnt: execute command (AT+CIPSTART=0,"TCP","192.168.1.87",1883,60) timeout (5000 ticks)!
<7>May 30 18:24:09 at.skt.esp32: esp0 device socket (0) connect failed, the socket was not be closed and now will connect retry.
<7>May 30 18:24:09 at.clnt: execute command (AT+CIPCLOSE=0) timeout (300 ticks)!
<3>May 30 18:24:09 mqtt: connect err!
<7>May 30 18:24:09 at.clnt: execute command (AT+CIPCLOSE=0) timeout (300 ticks)!
<3>May 30 18:24:09 mqtt: Net connect error(-2).
<7>May 30 18:24:09 mqtt: MQTT offline
<7>May 30 18:24:14 mqtt: restart!
<7>May 30 18:24:14 mqtt: MQTT connecting
<7>May 30 18:24:14 mqtt: ipv4 address port: 1883
<7>May 30 18:24:14 mqtt: HOST = '192.168.1.87'
<7>May 30 18:24:19 at.clnt: execute command (AT+CIPSTART=0,"TCP","192.168.1.87",1883,60) timeout (5000 ticks)!
<7>May 30 18:24:19 at.skt.esp32: esp0 device socket (0) connect failed, the socket was not be closed and now will connect retry.
<7>May 30 18:24:20 at.clnt: execute command (AT+CIPCLOSE=0) timeout (300 ticks)!
<3>May 30 18:24:20 mqtt: connect err!
<7>May 30 18:24:20 at.clnt: execute command (AT+CIPCLOSE=0) timeout (300 ticks)!
<3>May 30 18:24:20 mqtt: Net connect error(-2).
<7>May 30 18:24:20 mqtt: MQTT offline
<7>May 30 18:24:25 mqtt: restart!
<7>May 30 18:24:25 mqtt: MQTT connecting
<7>May 30 18:24:25 mqtt: ipv4 address port: 1883
<7>May 30 18:24:25 mqtt: HOST = '192.168.1.87'
<7>May 30 18:24:30 at.clnt: execute command (AT+CIPSTART=0,"TCP","192.168.1.87",1883,60) timeout (5000 ticks)!
<7>May 30 18:24:30 at.skt.esp32: esp0 device socket (0) connect failed, the socket was not be closed and now will connect retry.
<7>May 30 18:24:30 at.clnt: execute command (AT+CIPCLOSE=0) timeout (300 ticks)!
<3>May 30 18:24:30 mqtt: connect err!
<7>May 30 18:24:31 at.clnt: execute command (AT+CIPCLOSE=0) timeout (300 ticks)!
<3>May 30 18:24:31 mqtt: Net connect error(-2).
<7>May 30 18:24:31 mqtt: MQTT offline

```

ustccw, 2021-05-31:
@majianjia It looks like self-defined MQTT code broke the ESP-AT interval workflow.
Could you please share the project to us, and provide the steps for us to reproduce.

If possible, i would like to recommend the latest esp-at version, it got greatly improved at stability and compatibility.

majianjia, 2021-06-01:
@ustccw 
Thanks for your reply. I might locate the issue. It was the power. 
The PCB power supply was throught a 0.5mm FPC wire, which cannot provide enough current. 
I tested for a few hours now with a USB power supply. The issue was not replicated.

ustccw, 2021-06-03:
@majianjia Glad to hear that :smile:

closed 2021-06-21 by FayeY