espressif/esp-at #542: ESP8266烧录后不能正常启动

edieruby, 2021-05-31:
模组: ESP-WROOM-02D
ESP-AT:V2.1.0.0
烧录使用factory bin

[image]

烧录完成后启动日志为:
[image]

不设置DoNotChgBin, 尝试其他SPI参数, 都无法正常启动.

ustccw, 2021-05-31:
@edieruby please have a look: https://docs.espressif.com/projects/esp-at/zh_CN/latest/Get_Started/Downloading_guide.html

edieruby, 2021-06-02:
你好: 
   开发板和这个异常的模组, 烧录方式均相同, 使用的factory方式直接烧录, 开发板功能正常, 这个模组启动异常.
没有后续phy_version等日志. 
   没有其他错误日志,不好排查,  如果flash的datasheet不同,是否会有问题.

edieruby, 2021-06-02:
我重新买几个模组 再试试.

ustccw, 2021-06-03:
@edieruby What if erase flash before download.  
emmm, maybe you can measure the voltage between VCC and GND of ESP-WROOM-02 to make sure the power supply is OK.

edieruby, 2021-06-10:
供电问题, 串口供电不够

closed 2021-06-10 by edieruby