espressif/esp-at #544: WPA2 Enterprise, CWJEAP connect timeout. CA check

RomeliFeng, 2021-06-10:
Module: ESP32-WROOM-32

## Problem Description
Module can't not connect to Wi-Fi when radius server use self-signed certificate.

### Steps to repropduce
I have build firmware from the lastest commit and enable the WAP2 feature.
I used Unifi. UAP-AC-LITE and configured a WPA2 Enterprise Wi-Fi.

1. Wi-Fi radius server use unifi router. CA is UbiOS RADIUS Server Certificate.
Win10 computer can connect to Wi-Fi with simple connect, use username and password.
Module can connect to Wi-Fi too.

2. Wi-Fi radius server use Windows Server NPS. CA is domain self-signed certificate.
Win10 computer can connect to Wi-Fi with no CA check config.
Module can't connect to Wi-Fi. Error is timeout. (when i use esp32-arduino and enable debug info, the debug message is 23.802 1X AUTH_FAILED)

###
I think, the Root CA check caused this problem. Because Win10 can't connect to Wi-Fi with CA check too.

#### Help
How can I disable the CA check or add my own root CA to firmware.
I found infomation in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html .
CONFIG_ESP_TLS_INSECURE and CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY can use to skip server verification.
But in menuconfig, there have not CONFIG_ESP_TLS_INSECURE and CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY is RED can;t select.

Please help, thank you.

RomeliFeng, 2021-06-17:
Anyone help?

ustccw, 2021-06-21:
@RomeliFeng Hi, what if use the `wpa2 enterprise demo` of esp-idf?

btw, `wpa2 enterprise` is not based on `esp-tls` component. that is, `CONFIG_ESP_TLS_INSECURE` and `CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY` cfg would not affect the connection, whatever.

the whole source code of wpa2 enterprise could be found, under component `components/wpa_supplicant`.

closed 2021-07-01 by RomeliFeng