espressif/esp-at #546: Problems receiving files from AWS S3 bucket

charon22, 2021-06-17:
## Environment
- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32|
- AT version:2.2.0.0-dev(s-c7cd646 - ESP32 - May 20 2021 10:07:25)
  SDK version:v4.2.1-dirty
  compile time(c9c1fc4):Jun  4 2021 17:11:40
  Bin version:2.1.0(WROOM-32)
- Operating System:     [Windows]
- Power Supply:         [USB]
 
## Problem Description
* Problems receiving files from AWS S3 bucket
 
1. I am trying to receive a file from an AWS S3 bucket with the ESP-AT command.
2. When setting the file stored in the S3 bucket, we can create a URL that can be temporarily received with the ACCESS KEY.
3. An error occurs when trying to receive this URL using AT+HTTPCLIENT of the ESP-AT command.
    AT+HTTPCLIENT=2,0,"http://firmwaresbucket.s3.us-west-1.amazonaws.com/XXXX.bin?AWSAccessKeyId=XXXXXXX&Expires=1623969208&Signature=ZAe9Yz%2BwyOBc8ptXImn2JwFn9b4%3D","","",1

4. If the file is not temporarily saved and is set to public access, we can download the file.

I want to know how to download files from S3 bucket without public access.
-------------------------------------------------------------------------------------------------------------
Send AT Command :

AT+HTTPCLIENT=2,0,"http://firmwaresbucket.s3.us-west-1.amazonaws.com/XXXX.bin?AWSAccessKeyId=XXXXXXX&Expires=1623969208&Signature=ZAe9Yz%2BwyOBc8ptXImn2JwFn9b4%3D","","",1

### Expected Behavior
 AT+HTTPCLIENT=2,0,"https://qfw.s3-us-west-1.amazonaws.com/XXXX-2222.bin","","",1
+HTTPCLIENT:512,core hXCVCVXVC.......
+HTTPCLIENT:512,.....
+HTTPCLIENT:512,.....
+HTTPCLIENT:512,.....

### Actual Behavior
Response error message: 

 +HTTPCLIENT:241,<?xml version="1.0" encoding="UTF-8"?>
<Error><Code>SignatureDoesNotMatch</Code><Message>The request signature we calculated does not match the signature you provided. Check your key and signing method.</Message><AWSAccessKeyId>XXXXXXXX
+HTTPCLIENT:512,XXXXX</AWSAccessKeyId><StringToSign>GET

text/xml
1623969208
/firmwaresbucket/XXXX_bb2.bin</StringToSign><SignatureProvided>L3crQsSR+MJzU5ilIJvasA0AJCc=</SignatureProvided><StringToSignBytes>XX XX XX XX XX ....XX XX </StringToSignBytes><RequestId>K3X1BKGSQK6TK06H</RequestId><HostId>C7ZdKlSNlMP/KH0ccIjaZ6tvFmNHMnQNh16lV+6rz0/3Ye8sn8BuCzG
+HTTPCLIENT:38,ifVrdgbb3Kyf+arNMsyo=</HostId></Error>
-------------------------------------------------------------------------------------------------------------

### Steps to reproduce

1. step1 : upload file to AWS s3 bucket
2. step2 : Set it to not have public access.
3. step3 : Create a URL for temporary file access using java script.
  function : 
s3.getSignedUrl('getObject', {
    Bucket: bucket,
    Key: key,
    Expires: time
});
4. step4 : Attempt to download the file with AT+HTTPCLIENT command.
    AT+HTTPCLIENT=2,0,"http://firmwaresbucket.s3.us-west-1.amazonaws.com/XXXX.bin?AWSAccessKeyId=XXXXXXX&Expires=1623969208&Signature=ZAe9Yz%2BwyOBc8ptXImn2JwFn9b4%3D","","",1

ustccw, 2021-06-21:
@charon22 from the response u provide, it seems that the `SignatureDoesNotMatch`. that is, `url` parameter in `AT+HTTPCLIENT` may be incorrect. most likely, u need to escape some special characters. see the following for more details:
https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/index.html#at-command-types

charon22, 2021-06-21:
Hi ustccw,
For this problem, if you look at the log captured packets with wireshark, 
it looks like a normal request to the server. 
It doesn't seem to be a problem with the string character in the url parameter.
Before receiving this capture, I tested whether it was a character problem using an escape character, 
but the same problem occurred.

Thanks

ustccw, 2021-06-22:
what if use same url to visit aws s3 on pc web.  
if possible, compare the http request between the at http cmd and pc web, things would be clear.

charon22, 2021-06-22:
Hi ustccw,

Thanks for your advice.
As you said, there was no problem when I made a GET request to the same URL from the PC browser. 
I will send you the screenshot. 
[image]

Comparing the successful and unsuccessful cases, the difference is that there is no "Content-Type" item in the successful packet.
When I contacted AWS about this, they said that the "Contents-Type" was the problem. Below is the content of that answer.

--------------------------------------------------------------------------------------------------------------------------------------
If your request includes "Content-Type" header , then the value of this header must also be included while forming the signature or else the request would fail. I suspect , that while you are signing the request , the StringToSign does not include the value of the "Content-Type" header and hence when the request is made it fails due to signature mismatch error.
--------------------------------------------------------------------------------------------------------------------------------------

Currently, AT+HTTPCLIENT command cannot select anything without Content-Type.
I would like to know how to solve this problem.

Thanks

ustccw, 2021-06-23:
@charon22 there are two ways now.
### first one: remove `Content-Type` header in HTTP GET request
- replace the at-core with attachment: [libespat.zip](https://github.com/espressif/esp-at/files/6699707/libespat.zip)
- then, send the http command like: 
`AT+HTTPCLIENT=2,,"http://firmwaresbucket.s3.us-west-1.amazonaws.com/XXXX.bin?AWSAccessKeyId=XXXXXXX&Expires=1623969208&Signature=ZAe9Yz%2BwyOBc8ptXImn2JwFn9b4%3D",,,1`

It should work.

tbh, as the aws mentioned, esp-at should not send `Content-Type` header in http get request (it should be a bug).
so i extended the `<content-type>` parameter of `AT+HTTPCLIENT`: if it is omitted, then esp-at will not send the `Content-Type` header. plz test, esp-at will sync the bugfix to GitHub if it works.

### second one: use `Content-Type` and double-check signature
As the aws mentioned:
> If your request includes "Content-Type" header , then the value of this header must also be included while forming the signature or else the request would fail. I suspect , that while you are [...]

the thing you need to check is to inspect `StringToSign` and how the signature generated.

charon22, 2021-06-23:
Hi ustccw,

Thank you for your favorable and prompt reply.

** first one : remove Content-Type header in HTTP GET request
I tested with the command you sent. 
It's working fine.

** second one: use Content-Type and double-check signature
There are SIGv2 and SIGv4 presign versions supported for use in AWS S3 buckets. It is SIGv4 that requires StringToSign. 
I cannot input this parameters as the length limit of AT+HTTPCLIENT (256) because many parameters are required. 
In my case, only the length of the URL is 330.

> C:\AWS> aws s3 presign s3://firmwares/XXXXXXXX_bb2.bin --expires-in 300
https://firmwares.s3.us-west-1.amazonaws.com/XXXXXXXXXXX_bb2.bin?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=XXXXXXXXX120210623Fus-west-1XDFAEXFaws4_request&X-Amz-Date=20210623T172209Z&X-Amz-Expires=300&X-Amz-SignedHeaders=host&X-Amz-Signature=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

This is not a problem now, but it will become a problem in the future.
Thanks

ustccw, 2021-06-24:
@charon22 urw. first one got fixed on commit: https://github.com/espressif/esp-at/commit/625723aa62915fdb281ba687d175ea5531ebb57d

plz `git pull` latest repo, and test.

charon22, 2021-06-25:
Hi ustccw,
I tested this and it worked fine.
Thanks

closed 2021-06-25 by charon22