espressif/esp-at #548: When using Ethernet using the internal clock, it cannot get IP through DHCP.

charon22, 2021-06-28:
## Environment

- Development Kit:      |ESP32-DevKitC]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version:  v4.2.1-64-gd9ec7df39
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
I want to connect Ethernet using LAN8720, and connect to supply clock to LAN8720 using ESP internal clock.
If you connect the Ethernet line, you cannot get an IP after "starting dhcp client".
The MAC address is received normally.
When confirmed with AT+CIFSR command,  the ethernet IP is 0.0.0.0.

When I supplied RMII using an external clock, there was no problem in using WIFI and Ethernet at the same time.

### Expected Behavior
 there was no problem in using WIFI and Ethernet at the same time.
ready
WIFI CONNECTED
WIFI GOT IP
+ETH_CONNECTED
+ETH_GOT_IP:10.0.28.173
at

OK
AT+CIFSR
+CIFSR:APIP,"192.168.4.1"
+CIFSR:APMAC,"24:0a:c4:f9:cf:c5"
+CIFSR:STAIP,"10.0.28.165"
+CIFSR:STAMAC,"24:0a:c4:f9:cf:c4"
+CIFSR:ETHIP,"10.0.28.173"
+CIFSR:ETHMAC,"24:0a:c4:f9:cf:c7"

### Actual Behavior
ready
+ETH_CONNECTED
AT+CIFSR
+CIFSR:APIP,"192.168.4.1"
+CIFSR:APMAC,"9c:9c:1f:c6:03:59"
+CIFSR:ETHIP,"0.0.0.0"
+CIFSR:ETHMAC,"9c:9c:1f:c6:03:5b"

OK
### Steps to repropduce

1. step1 : Remove the oscillator connection from the LAN8720 development board.
2. step 2 : Connect GPIO5 to reset on LAN8720.
3. step 3 : Connect to ethernet cable
4. step 4 : Confirm IP address

## Debug Logs
[teraterm_COM15_20210628.122228_intenal_clock_problem.log](https://github.com/espressif/esp-at/files/6729686/teraterm_COM15_20210628.122228_intenal_clock_problem.log)
D (3323) esp_eth.phy: Found PHY address: 1
**I (3346) esp_eth.netif.glue: **9c:9c:1f:c6:03:5b****
I (3346) esp_eth.netif.glue: ethernet attached to netif
D (3357) event: running post ETH_EVENT:0 with handler 0x40148b4c and context 0x3ffd6828 on loop 0x3ffbb00c
D (3357) esp_netif_handlers: esp_netif action has started with netif0x3ffd67a0 from event_id=0
D (3364) esp_netif_lwip: check: remote, if=0x3ffd67a0 fn=0x40149dd8

D (3371) esp_netif_lwip: esp_netif_start_api 0x3ffd67a0
D (3376) esp_netif_lwip: esp_netif_get_hostname esp_netif:0x3ffd67a0
D (3383) esp_netif_lwip: check: local, if=0x3ffd67a0 fn=0x4014aa80

D (3389) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffd67a0
V (3396) esp_netif_objects: esp_netif_next_unsafe 0
V (3400) esp_netif_objects: esp_netif_next_unsafe 0x3ffd67a0
V (3406) esp_netif_objects: esp_netif_next_unsafe 0x3ffd68b8
V (3412) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbdaa4
V (3418) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffd67a0
D (3424) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (3430) event: running post ETH_EVENT:0 with handler 0x400e38a8 and context 0x3ffb55a8 on loop 0x3ffbb00c
D (7357) emac_esp32: working in 100Mbps
D (7357) emac_esp32: working in full duplex
D (7358) event: running post ETH_EVENT:2 with handler 0x40148b9c and context 0x3ffd6878 on loop 0x3ffbb00c
D (7364) esp_netif_handlers: esp_netif action connected with netif0x3ffd67a0 from event_id=2
D (7372) esp_netif_lwip: check: remote, if=0x3ffd67a0 fn=0x40149fcc

D (7379) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffd67a0
D (7385) esp_netif_lwip: check: local, if=0x3ffd67a0 fn=0x4014aa80

D (7391) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffd67a0
V (7398) esp_netif_objects: esp_netif_next_unsafe 0
V (7403) esp_netif_objects: esp_netif_next_unsafe 0x3ffd67a0
V (7408) esp_netif_objects: esp_netif_next_unsafe 0x3ffd68b8
V (7414) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbdaa4
V (7420) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffd67a0
D (7426) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (7432) esp_netif_lwip: check: remote, if=0x3ffd67a0 fn=0x401498e0

D (7439) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffd67a0
D (7445) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffd67a0
D (7452) esp_netif_lwip: if0x3ffd67a0 start ip lost tmr: no need start because netif=0x3ffd6af8 interval=120 ip=0
D (7463) esp_netif_lwip: starting dhcp client
D (7468) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (7473) event: running post ETH_EVENT:2 with handler 0x400e38a8 and context 0x3ffb55a8 on loop 0x3ffbb00c
D (7957) emac_esp32: receive len= 62
D (8297) emac_esp32: receive len= 60
D (9515) emac_esp32: receive len= 60
D (9593) emac_esp32: receive len= 60

## SDK Config
[sdkconfig.log](https://github.com/espressif/esp-at/files/6729690/sdkconfig.log)

charon22, 2021-06-30:
* There were several symptoms of this problem.
1. Not recognizing PHY
> [2021-06-29 13:41:25.055] V (3309) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
[2021-06-29 13:41:25.063] V (3315) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
[2021-06-29 13:41:25.068] D (3323) intr_alloc: Connected src 38 to int 17 (cpu 0)
[2021-06-29 13:41:25.073] E (3331) esp_eth.phy: No PHY device detected
[2021-06-29 13:41:25.083] E (3333) lan8720: lan8720_init(380): Detect PHY address failed
[2021-06-29 13:41:25.085] E (3339) esp_eth: esp_eth_driver_install(199): init phy failed
[2021-06-29 13:41:25.092] regist ethernet cmd fail
2. Cannot get IP through DHCP
> D (7463) esp_netif_lwip: starting dhcp client
D (7468) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (7473) event: running post ETH_EVENT:2 with handler 0x400e38a8 and context 0x3ffb55a8 on loop 0x3ffbb00c
D (7957) emac_esp32: receive len= 62
D (8297) emac_esp32: receive len= 60
D (9515) emac_esp32: receive len= 60

* This test was mainly performed by reconnecting the USB of the development board.
* Now, after resetting with the EN button on the development board, there was no problem.
* But I still have doubts about this. There seems to be a problem with the order of connecting the power.
* I would like to know why the button behaves differently than when it is pressed. 
* Like the problem raised in the link below, the clock must be enabled first when booting the device, but GPIO5 seems to be enabled first. We will also send you a capture of the signal.
  > https://github.com/espressif/esp-idf/issues/6821
[image]

Thanks

ustccw, 2021-07-05:
@charon22 i am trying to apply for a development board that supports LAN8720. Once I have the board, I will check it as soon as possible, and then give the feedback to you here.

charon22, 2021-07-08:
Hi ESP-AT,

We compared the signal when the USB power is reconnected and the EN button is pressed.
1. USB Power is reconnected
[image]
  > After 1)reset, before the 2)rx signals go low, 3)the PHY address is requested. In this case, an error occurs.

2. EN button is pressed
[image]
> After reset, the rx signals go low almost at the same time, and then the PHY address is requested. In this case, the PHY address is successfully obtained.

Comparing the above two cases, when the USB power is reconnected, the RX signal changes from High to Low very late. As a result, it seems that the PHY address cannot be obtained.

I would like to make a few suggestions on this.
> 1. When initializing PHY, put a delay of about 15msec before obtaining the PHY address.
> 2. Retry when getting PHYaddress fails
Below is the related example code.

> static esp_err_t lan8720_init(esp_eth_phy_t *phy)
> {
>     phy_lan8720_t *lan8720 = __containerof(phy, phy_lan8720_t, parent);
>     esp_eth_mediator_t *eth = lan8720->eth;
>     // Detect PHY address
>     if (lan8720->addr == ESP_ETH_PHY_ADDR_AUTO) {
> 		ets_delay_us(15000); // insert min input assert time
> 		//PHY_CHECK(esp_eth_detect_phy_addr(eth, &lan8720->addr) == ESP_OK, "Detect PHY address failed", err); // org
> 		int checkcnt = 0;
> 		for ( ; checkcnt < 10 ; checkcnt++ )
> 		{
>         	if ( esp_eth_detect_phy_addr(eth, &lan8720->addr) == ESP_OK)
> 				break;
> 			ESP_LOGE(TAG, "%s(%d): Detect PHY address failed(%d)", __FUNCTION__, __LINE__,checkcnt);
> 			if ( checkcnt == 9)
> 			{
> 				goto err;
> 			}
> 		}
>     }
>     /* Power on Ethernet PHY */
>     PHY_CHECK(lan8720_pwrctl(phy, true) == ESP_OK, "power control failed", err);
>     /* Reset Ethernet PHY */
>     PHY_CHECK(lan8720_reset(phy) == ESP_OK, "reset failed", err);
>     /* Check PHY ID */
>     phyidr1_reg_t id1;
>     phyidr2_reg_t id2;
>     PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_IDR1_REG_ADDR, &(id1.val)) == ESP_OK,
>               "read ID1 failed", err);
>     PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_IDR2_REG_ADDR, &(id2.val)) == ESP_OK,
>               "read ID2 failed", err);
>     PHY_CHECK(id1.oui_msb == 0x7 && id2.oui_lsb == 0x30 && id2.vendor_model == 0xF, "wrong chip ID", err);
>     return ESP_OK;
> err:
>     return ESP_FAIL;
> }
This is the log captured when this code is applied. Even if the RX signal goes low late, we can get PHY address later.

[image]

Below is the result of testing with the 15msec delay removed from the code above, leaving only the retry code. On the second attempt, the PHY value is normally obtained. 
However, there is a significant delay between the first and second request times, and I am curious as to why this is so.
[image]

I'm not sure what caused this problem and which one to apply.
I would like to hear your comments on this.

I think this is an ESP-IDF problem, but I have to use ESP-AT now, so I'm posting this here again. I'd like to share your opinion on this.

Thanks.

ustccw, 2021-07-12:
@charon22 Thanks for the detailed explanation. i got an LAN8720 today and test. it looks everything works well with esp-at.

[image]

---
So firstly, Could you please test the ethernet demo in esp-idf: 
https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic
If everything is okey with your environment, you should get the log like the following: (**it is necessary for you to run esp-at based on ethernet**)
```
I (26) boot: ESP-IDF v4.2.2-99-g96406ea6a5 2nd stage bootloader
I (26) boot: compile time 17:37:23
I (27) boot: chip revision: 0
I (31) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0d850 ( 55376) map
I (117) esp_image: segment 1: paddr=0x0001d878 vaddr=0x3ffb0000 size=0x020d4 (  8404) load
I (121) esp_image: segment 2: paddr=0x0001f954 vaddr=0x40080000 size=0x006c4 (  1732) load
0x40080000: _WindowOverflow4 at /home/chenwu/esp/esp-idf-v4.2/components/freertos/xtensa/xtensa_vectors.S:1730

I (125) esp_image: segment 3: paddr=0x00020020 vaddr=0x400d0020 size=0x2dc84 (187524) map
0x400d0020: _stext at ??:?

I (204) esp_image: segment 4: paddr=0x0004dcac vaddr=0x400806c4 size=0x0a1f8 ( 41464) load
I (229) boot: Loaded app from partition at offset 0x10000
I (229) boot: Disabling RNG early entropy source...
I (229) cpu_start: Pro cpu up.
I (233) cpu_start: Application information:
I (238) cpu_start: Project name:     ethernet_basic
I (243) cpu_start: App version:      v4.2.2-99-g96406ea6a5
I (249) cpu_start: Compile time:     Jul 12 2021 18:14:21
I (255) cpu_start: ELF file SHA256:  e5dcf5b8180d84d1...
I (261) cpu_start: ESP-IDF:          v4.2.2-99-g96406ea6a5
I (268) cpu_start: Starting app cpu, entry point is 0x4008174c
0x4008174c: call_start_cpu1 at /home/chenwu/esp/esp-idf-v4.2/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (278) heap_init: Initializing. RAM available for dynamic allocation:
I (285) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (291) heap_init: At 3FFB3868 len 0002C798 (177 KiB): DRAM
I (297) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (303) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (310) heap_init: At 4008A8BC len 00015744 (85 KiB): IRAM
I (316) cpu_start: Pro cpu start user code
I (334) spi_flash: detected chip: gd
I (335) spi_flash: flash io: dio
W (335) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (345) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (356) system_api: Base MAC address is not set
I (366) system_api: read default base MAC address from EFUSE
I (386) esp_eth.netif.glue: 30:ae:a4:80:59:cb
I (386) esp_eth.netif.glue: ethernet attached to netif
I (1886) eth_example: Ethernet Started
I (1886) eth_example: Ethernet Link Up
I (1886) eth_example: Ethernet HW Addr 30:ae:a4:80:59:cb
I (4356) esp_netif_handlers: eth ip: 192.168.200.32, mask: 255.255.255.0, gw: 192.168.200.1
I (4356) eth_example: Ethernet Got IP Address
I (4356) eth_example: ~~~~~~~~~~~
I (4356) eth_example: ETHIP:192.168.200.32
I (4366) eth_example: ETHMASK:255.255.255.0
I (4366) eth_example: ETHGW:192.168.200.1
I (4376) eth_example: ~~~~~~~~~~~

```

Then Secondly, the only thing you should mind about esp-at based on ethernet is that:
`LAN8720` `RETCLK pin` must connect to GPIO17 of ESP32. so there is a pin conflict. (esp-at uart1 use GPIO17 and GPIO16 by default). You could set another pins for uart1 according to https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_set_AT_port_pin.html#esp32-series

Thanks.

FayeY, 2021-07-30:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details. Thanks.

closed 2021-07-30 by FayeY

charon22, 2021-07-31:
Hi ESP-AT,

We tested it as you said.
Two problems occurred during testing.

**First**, I set GPIO 17 as internal clock, and modified GPIO for AT command to 9 and 10 (ESP UART1).
When I tested using Teraterm with GPIOs 9 and 10, there seemed to be no major problems.
But, when connected to the device, a strange signal(high frequency signal) was sent from ESP's UART1 TX.
Because of that, our board's RX has a problem and can't be used.
uart signal is usually 8-10 usec, but a strange signal is 1 usec.

**Second**, I tried to use ESP's UART0 and UART1 interchangeably. 
When AT command was used with UART0 (GPIO 1,3), it worked without any problem. 
But, a problem occurred when trying to download the firmware in this state. 
I tried multiple times, but failed to download the firmware.

Thanks,