espressif/esp-at #549: Build Error on Windows 10

hattmt, 2021-07-06:
Hello !

I tried to build ESP-AT v2.2 with ESP-IDF 4.2 on windows 10,  for ESP32-WROOM 
but at the end I get a git tree error.

[image]

I succeded before with previous version on Debian but with the new release it was stated that we could build it on Windows.
So I tried but I failed.
Do you have a solution for this problem ?

Regards.

charon22, 2021-07-06:
Hi ESP-AT,
I also tried to build a new ESP-AT development environment on another computer.
The same error as above occurred.

So when I tried to delete the existing project from my computer and get it again, the above error occurred.

SquaredPotato, 2021-07-13:
I'm having the same error on MacOS, oddly enough.
Also tried a fresh download and I got the same error.

ezlife, 2021-07-17:
modify esp-at\module_config\module_esp32_default\IDF_VERSION commit with new
commit:1e3638390964c86b0e7ef3af02973b30c58ea1be
repository:https://github.com/espressif/esp-idf.git

ustccw, 2021-07-19:
@hattmt @charon22 @SquaredPotato @ezlife So sorry for that. But it should work now.
> This is because esp-idf release/v4.2 cannot sync to GitHub automatically before. we pushed it internally, and now it is available now.

please try again.

SquaredPotato, 2021-07-19:
This issue is resolved for me now using the master branch, thanks!

hattmt, 2021-07-20:
The build also worked for me, thanks !

closed 2021-07-30 by FayeY