espressif/esp-at #551: ESP-AT Ethernet 運作起來後 , DHCP server 是有辦法啟動嗎?

adiogithub, 2021-07-10:
ESP-AT Ethernet 編譯好後. 燒入到我們設計的硬件板子上.

經過測試 dhcp client 運作正常.
+ETH_CONNECTED
+ETH_GOT_IP:10.1.1.100
+ETH_DISCONNECTED

問題 1 :
但我找不到 at 指令可以啟用 dhcp server.
以滿足以下使用情境 : 
我的筆記型電腦 RJ45 接上 ESP-AT Ethernet 的 RJ45 .
但筆記型電腦並沒有取得任何的 IP address.

問題2:
並且如果你使用 AT+CIPETH="192.168.6.100","192.168.6.1","255.255.255.0"
之後 再也不會運作起來 dhcp client. 也就是 該 ESP-AT Ethernet 永遠都是 192.168.6.100 的 IP address.

ustccw, 2021-07-19:
@adiogithub
Q1: How to enable ethernet dhcp server.
A1: esp-at does not support ethernet dhcp server now, just support ethernet dhcp client.
A1: it is recommended that use self-defined AT commands if it is urgent for you. More details: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_add_user-defined_AT_commands.html
we should consider this feature at internally at the same time.

Q2: will check it this week.

ustccw, 2021-07-23:
@adiogithub About Q2, i tested it and it seemed to work well. The PHY of Ethernet I use is IP101 PHY, which is built into the [ESP32-Ethernet-Kit](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html) development board.

[image]

For more details, i would like to suggest you enable the debug log in menuconfig to see the specific reasons.

adiogithub, 2021-07-24:
Hi ,

你好 , 請問能提供 esp32 使用 RTL8201 參考線圖嗎?

BR,
polk.

ustccw, 2021-07-29:
@adiogithub Could you please help to test the ethernet demo: https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic, see if the demo works well.

closed 2023-06-05 by ustccw