espressif/esp-at #553: ESP-AT Build failed, because of missing ble_data.bin

Caffreyfans, 2021-07-22:
## Environment

- Development Kit:  ``ESP32-WROOM-32``
- IDF version :  ``v2.2.0.0_esp32c3-63-gca2654c``
- Build System:  ``CMake``
- Compiler version:  ``(crosstool-NG esp-2020r3) 8.4.0``
- Operating System: ``Linux``
- Power Supply:  ``USB``

## Problem Description

Follow the tutorial to compile ESP-AT project failed. [Tutorial Link](https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html)

### Steps to repropduce

1. step1
```shell
git clone --recursive https://github.com/espressif/esp-at.git
```
2. step 2
```shell
python -m pip install pyyaml xlrd
```

3. step 3
```shell
cd esp-at && ./build.py menuconfig
```
4. step4
```shell
./build.py build
```

Output:
```shell
Scanning dependencies of target factory_bin
Create factory bin...
0x8000,partition_table/partition-table.bin
0x10000,ota_data_initial.bin
0xf000,phy_init_data.bin
0x1000,bootloader/bootloader.bin
0x100000,esp-at.bin
0x20000,at_customize.bin
0x21000,customized_partitions/ble_data.bin
Traceback (most recent call last):
  File "/home/caffreyfans/git/esp-at/tools/esp_at_factory_bin_combine.py", line 98, in <module>
    main()
  File "/home/caffreyfans/git/esp-at/tools/esp_at_factory_bin_combine.py", line 94, in main
    esp32_at_combine_bin(args.module_name.upper(), args.flash_mode.upper(), args.flash_size.upper(),
  File "/home/caffreyfans/git/esp-at/tools/esp_at_factory_bin_combine.py", line 47, in esp32_at_combine_bin
    with open(os.path.join(build_dir, bin_file), 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/caffreyfans/git/esp-at/build/customized_partitions/ble_data.bin'
make[2]: *** [CMakeFiles/factory_bin.dir/build.make:62: factory_bin] Error 1
make[1]: *** [CMakeFiles/Makefile2:1978: CMakeFiles/factory_bin.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2
idf.py build ret: 2
idf.py build failed
```

---------------------------------------------------------------------------------------------------

This seems to be missing the ``ble_data.bin`` file, but I don't know how to deal with it. Maybe somebody can save me.

BRs
Caffreyfans

Caffreyfans, 2021-07-22:
@xiewenxiang
@ustccw

Please have a look.

ustccw, 2021-07-23:
@Caffreyfans please `rm -rf build sdkconfig` and compile again, save the complete log.

i guess it might be a component installation failure. Emm, double check the `python -m pip install pyyaml xlrd` returned.

Caffreyfans, 2021-07-27:
@ustccw Thanks a lot, the reason is missing ```pyyaml``` module.

closed 2021-07-27 by Caffreyfans