espressif/esp-at #554: esp8266 AT+MQTTUSERCFG always error

Funqiewei, 2021-07-29:
ready
WIFI CONNECTED
WIFI GOT IP
AT+CIPSTA?
+CIPSTA:ip:"172.20.10.4"
+CIPSTA:gateway:"172.20.10.1"
+CIPSTA:netmask:"255.255.255.240"

OK
AT+CWMODE?
+CWMODE:1

OK
AT+CWJAP="anchovy","1234567890"

OK
AT+CIPSNTPCFG=1,8,"ntp1.aliyun.com","ntp2.aliyun.com"

OK
AT+MQTTUSERCFG=0,5,"ESP32","espressif","1234567890",0,0,""

ERROR

Funqiewei, 2021-07-29:
AT+MQTTUSERCFG=0,1,"ESP8266","espressif","1234567890",0,0,""
still keep error

ustccw, 2021-07-29:
@Funqiewei please provide the [AT+GMR](https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/AT_Command_Set/Basic_AT_Commands.html#cmd-gmr) and [AT log port output](https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/Get_Started/Hardware_connection.html#esp8266-series).

Funqiewei, 2021-07-30:
@ustccw   sorry ,i dont know how to provide the AT log
Here is AT+GMR
WIFI CONNECTED
WIFI GOT IP
AT+GMR
AT version:1.7.4.0(May 11 2020 19:13:04)
SDK version:3.0.4(9532ceb)
compile time:May 27 2020 10:12:17
Bin version(Wroom 02):1.7.4
OK

ustccw, 2021-07-30:
@Funqiewei `v1.7.4.0` AT firmware did not support `MQTT` commands.

Please download the `v2.2.0.0` AT firmware here: https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/AT_Binary_Lists/ESP8266_AT_binaries.html

Of course, user can self-compile the `ESP-AT` project according to https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/Compile_and_Develop/How_to_clone_project_and_compile_it.html

closed 2021-09-24 by Funqiewei