espressif/esp-at #555: LAN8920 Problem

charon22, 2021-08-02:
## Environment
- https://github.com/espressif/esp-at/issues/548

Hi ESP-AT,
I couldn't reopen the issue for the following issues, so I made an issue again.

We tested it as you said.
Two problems occurred during testing.

First, I set GPIO 17 as internal clock, and modified GPIO for AT command to 9 and 10 (ESP UART1).
When I tested using Teraterm with GPIOs 9 and 10, there seemed to be no major problems.
But, when connected to the device, a strange signal(high frequency signal) was sent from ESP's UART1 TX.
Because of that, our board's RX has a problem and can't be used.
uart signal is usually 8-10 usec, but a strange signal is 1 usec.

Second, I tried to use ESP's UART0 and UART1 interchangeably.
When AT command was used with UART0 (GPIO 1,3), it worked without any problem.
But, a problem occurred when trying to download the firmware in this state.
I tried multiple times, but failed to download the firmware.

Thanks,

ustccw, 2021-08-02:
@charon22 GPIO9 and GPIO10 is used for communicate with flash, it should not be used for UART1.

Firmware download is always by UART0. could you please provide the download log in failed case?

How about UART0 and UART1 use same GPIOs (GPIO1 as TX, GPIO3 as RX) ? 
https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_set_AT_port_pin.html

charon22, 2021-08-03:
Hi ustccw,

Problem when using ESP internal clock when using ethernet PHY evaluation board lan8920 in ESP-AT.

> 1. When using UART2(GPIO 16,17) of ESP as ESP-AT command, and when using GPIO0 as REF_CLK of RMII, there was a problem with the clock. So, I solved it by extending the RMII clock line. This is difficult to solve in H/W when making a PCB board, so REF_CLK of RMII is modified to GPIO17.
[image]
** Question**  If we use UART2 as UART port for ESP-AT command, we want to know why the line of RMII REF_CLK has to be doubled. How can I solve this problem?

> 2. When using UART1(GPIO 9,10) of ESP as ESP-AT command, and when using GPIO17 as REF_CLK of RMII, there was no problem with the clock. But, A problem occurred when using the ESP-AT command. When using as UART1, abnormal signals of 1us sometimes come in, making UART communication impossible. See below for more details on this. 
[image]
[image]
[2021-07-30_17-18-26-2.zip](https://github.com/espressif/esp-at/files/6926861/2021-07-30_17-18-26-2.zip)
> A normal UART signal has a period of 8.6us, but an abnormal signal has a period of 1.6 us, making UART unusable.
**Question**  If we want to know how to cancel the signal such as noise generated when we use UART1 as UART port for ESP-AT command. How can I solve this problem? 
=> If method 1) does not solve the problem, method 2) has to be used.

**Question** 
I want to know how can we use UART0 as UART port for ESP-AT command ? .
If the above two methods do not solve the problem, you have no choice but to use UART0 as the ESP-AT command.

Thanks

ustccw, 2021-08-09:
@charon22 just a little confused about your hardware, `LAN8920` or `LAN8720`?.
Have you tested the esp-idf ethernet demo, https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic.

In your case, i would like to suggest you write a test demo based on your hardware and submit it to the [esp-idf issues](https://github.com/espressif/esp-idf/issues).

charon22, 2021-08-09:
Hi ustccw,
I'm using LAN8720, so I'll ask the ESP-IDF question.
Thanks

closed 2023-06-05 by ustccw