espressif/esp-at #561: 无法调用esp_at_port_write_data();

AHulyi, 2021-09-03:
我现在在用esp-at做二次开发，在esp-at中添加了我自己代码myproject，代码结构大致如下：
|---eap-at
	|---components
		|---at
			|---include
				|---esp_at_core.h
				|---eap_at.h
	|---main
	|---myproject
                |---CMakeLists.txt
		|---driver
			|---dev.c
	|---CMakeLists.txt
现在我需要在 dev.c 文件中调用esp_at_port_write_data();来将一些信息从AT指令的端口输出，在dev.c文件中我添加了#include "esp_at_core.h"文件,编译后出现：
    fatal error: esp_at_core.h: No such file or directory
     #include "esp_at_core.h"
我尝试在myproject/CMakeLists.txt中添加：
1、
    INCLUDE_DIRECTORIES($ENV{ESP_AT_PROJECT_PATH}/components/at/include)

2、
idf_component_register(SRCS ${srcs}
                 INCLUDE_DIRS  "."
                 REQUIRES components
)

但是还是不可以。
请问我需要怎样做才能在我的代码中调用此API？感谢解答！

ustccw, 2021-09-07:
@AHulyi 
1. 目录结构似乎都是平级的
2. 可以多加一些打印信息在 cmake 里，例如：`message("ESP-AT Path:$ENV{ESP_AT_PROJECT_PATH}")`
3. 也可以现在文件里用绝对路径测试看看，例如：`#include "/home/ahulyi/esp/esp-at/components/at/include/esp_at_core.h"`

AHulyi, 2021-09-07:
多谢回复，我粘贴目录结构的时候忘了调整，已经解决了这个问题，我将我的工程放在components下，并在我的工程中的CMakeLists.txt加了at组件，可以调用相应的函数。

closed 2021-09-07 by AHulyi