espressif/esp-at #563: AT+BLEINIT=2 ERROR

AntonDrubetskiy, 2021-09-09:
Hey!
I am using ESP32-C3 on DevKitC-02 board. I am trying to initialize BLE via AT + BLEINIT = 2 and I get this response

AT + BLEINIT = 2
AT + BLEINIT = 2

ERROR

At the same time, the following message appears on the UART as a channel for the logs

Config struct mismatch: magic = 5a5aa5a5, ver = 02103310

The steps described in the official documentation for enabling the BLE server I did https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/BLE_AT_Commands.html#cmd-binit
In addition, I noticed that the latest version of ESP-AT itself generates the necessary files and flashes them to the desired addresses in the flash. I used the AT + SYSFLASH? Command to check. She gave me such a log

AT + SYSFLASH?
AT + SYSFLASH?
+ SYSFLASH: "ble_data", 64,1,0x1f000,0x6000
+ SYSFLASH: "server_cert", 64,2,0x25000,0x2000
+ SYSFLASH: "server_key", 64,3,0x27000,0x2000
+ SYSFLASH: "server_ca", 64,4,0x29000,0x2000
+ SYSFLASH: "client_cert", 64,5,0x2b000,0x2000
+ SYSFLASH: "client_key", 64,6,0x2d000,0x2000
+ SYSFLASH: "client_ca", 64,7,0x2f000,0x2000
+ SYSFLASH: "factory_param", 64,8,0x31000,0x1000
+ SYSFLASH: "wpa2_cert", 64,9,0x32000,0x2000
+ SYSFLASH: "wpa2_key", 64,10,0x34000,0x2000
+ SYSFLASH: "wpa2_ca", 64,11,0x36000,0x2000
+ SYSFLASH: "mqtt_cert", 64,12,0x38000,0x2000
+ SYSFLASH: "mqtt_key", 64,13,0x3a000,0x2000
+ SYSFLASH: "mqtt_ca", 64,14,0x3c000,0x2000
+ SYSFLASH: "fatfs", 1,129,0x47000,0x19000

OK

Tell me what am I doing wrong? Thank you in advance!

SUNLUXIN, 2021-09-10:
@AntonDrubetskiy In response to your issue, can you do the following two operations to let us know more about the issue?
1. please provide the AT+GMR and AT log port output.
2. you can try it with the firmware in the attachment below to see if the same issue.

SUNLUXIN, 2021-09-10:
@AntonDrubetskiy In response to your issue, can you do the following two operations to let us know more about the issue?

1. please provide the AT+GMR and AT log port output.
2. you can try it with the firmware in the attachment below to see if the same issue.
[C3_MASTER.tar.gz](https://github.com/espressif/esp-at/files/7141700/C3_MASTER.tar.gz)

AntonDrubetskiy, 2021-09-10:
> @AntonDrubetskiy In response to your issue, can you do the following two operations to let us know more about the issue?
>
> 1. please provide the AT+GMR and AT log port output.
> 2. you can try it with [...]

Thank you very much!)) The firmware C3_MASTER.tar.gz is working!))

AntonDrubetskiy, 2021-09-10:
Just in case, I will leave the logs of a working and non-working firmware

**Working firmware** AT + GMR:

AT + GMR
AT + GMR
AT version: 2.3.0.0-dev (5092dec - ESP32C3 - Sep 3 2021 12:00:11)
SDK version: qa-test-v4.3.1-20210705-67-g47b
compile time (e2b9d35): Sep 8 2021 10:42:51
Bin version: 2.2.0 (MINI-1)

OK

**Working firmware** AT log on restart:

ESP-ROM: esp32c3-api1-20210207
Build: Feb 7 2021
rst: 0x1 (POWERON), boot: 0xc (SPI_FAST_FLASH_BOOT)
SPIWP: 0xee
mode: DIO, clock div: 2
load: 0x3fcd6100, len: 0x17d4
load: 0x403ce000, len: 0x8dc
load: 0x403d0000, len: 0x2984
entry 0x403ce000
I (30) boot: ESP-IDF qa-test-v4.3.1-20210705-67-g47b 2nd stage bootloader
I (31) boot: compile time 10:41:33
I (31) boot: chip revision: 3
I (35) boot.esp32c3: SPI Speed: 40MHz
I (39) boot.esp32c3: SPI Mode: DIO
I (44) boot.esp32c3: SPI Flash Size: 4MB
I (49) boot: Enabling RNG early entropy source ...
I (54) boot: Partition Table:
I (58) boot: ## Label Usage Type ST Offset Length
I (65) boot: 0 otadata OTA data 01 00 0000d000 00002000
I (72) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (80) boot: 2 nvs WiFi data 01 02 00010000 0000e000
I (87) boot: 3 at_customize unknown 40 00 0001e000 00042000
I (95) boot: 4 ota_0 OTA app 00 10 00060000 001d0000
I (102) boot: 5 ota_1 OTA app 00 11 00230000 001d0000
I (110) boot: End of partition table
I (114) esp_image: segment 0: paddr = 00060020 vaddr = 3c130020 size = 28200h (164352) map
I (157) esp_image: segment 1: paddr = 00088228 vaddr = 3fc8fc00 size = 03adch (15068) load
I (161) esp_image: segment 2: paddr = 0008bd0c vaddr = 40380000 size = 0430ch (17164) load
I (167) esp_image: segment 3: paddr = 00090020 vaddr = 42000020 size = 12c428h (1229864) map
I (429) esp_image: segment 4: paddr = 001bc450 vaddr = 4038430c size = 0b7e0h (47072) load
I (440) esp_image: segment 5: paddr = 001c7c38 vaddr = 50,000,000 size = 00004h (4) load
I (441) esp_image: segment 6: paddr = 001c7c44 vaddr = 50000008 size = 00010h (16) load
I (451) boot: Loaded app from partition at offset 0x60000
I (452) boot: Disabling RNG early entropy source ...
module_name: MINI-1
max tx power = 78, ret = 0
2.2.0

Just in case, I will leave the logs of a working and non-working firmware

**Not working** AT + GMR firmware:

AT + GMR
AT + GMR
AT version: 2.3.0.0-dev (5092dec - ESP32C3 - Sep 3 2021 12:00:11)
SDK version: v4.3
compile time (4a9961d): Sep 9 2021 10:25:00
Bin version: 2.2.0 (MINI-1)

OK

**Not working firmware** AT log on restart:

ESP-ROM: esp32c3-api1-20210207
Build: Feb 7 2021
rst: 0x1 (POWERON), boot: 0xc (SPI_FAST_FLASH_BOOT)
SPIWP: 0xee
mode: DIO, clock div: 2
load: 0x3fcd6100, len: 0x1774
load: 0x403ce000, len: 0x8d4
load: 0x403d0000, len: 0x293c
entry 0x403ce000
I (30) boot: ESP-IDF v4.3 2nd stage bootloader
I (30) boot: compile time 14:56:25
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed: 40MHz
I (37) boot.esp32c3: SPI Mode: DIO
I (41) boot.esp32c3: SPI Flash Size: 4MB
I (46) boot: Enabling RNG early entropy source ...
I (52) boot: Partition Table:
I (55) boot: ## Label Usage Type ST Offset Length
I (62) boot: 0 otadata OTA data 01 00 0000d000 00002000
I (70) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (77) boot: 2 nvs WiFi data 01 02 00010000 0000e000
I (85) boot: 3 at_customize unknown 40 00 0001e000 00042000
I (92) boot: 4 ota_0 OTA app 00 10 00060000 001d0000
I (100) boot: 5 ota_1 OTA app 00 11 00230000 001d0000
I (107) boot: End of partition table
I (112) esp_image: segment 0: paddr = 00060020 vaddr = 3c140020 size = 2b048h (176200) map
I (158) esp_image: segment 1: paddr = 0008b070 vaddr = 3fc8f800 size = 03d14h (15636) load
I (162) esp_image: segment 2: paddr = 0008ed8c vaddr = 40380000 size = 0128ch (4748) load
I (165) esp_image: segment 3: paddr = 00090020 vaddr = 42000020 size = 1379f8h (1276408) map
I (447) esp_image: segment 4: paddr = 001c7a20 vaddr = 4038128c size = 0e37ch (58236) load
I (462) esp_image: segment 5: paddr = 001d5da4 vaddr = 50,000,000 size = 00004h (4) load
I (462) esp_image: segment 6: paddr = 001d5db0 vaddr = 50000008 size = 00010h (16) load
I (472) boot: Loaded app from partition at offset 0x60000
I (473) boot: Disabling RNG early entropy source ...
module_name: MINI-1
max tx power = 78, ret = 0
2.2.0

closed 2021-09-10 by AntonDrubetskiy

Luxin812, 2021-09-10:
@AntonDrubetskiy The C3_MASTER firmware is only for testing, not a release version.
I saw the SDK version: v4.3 version you used to compile AT from the non-working log you provided above. This may be the cause of this issue. In order to avoid this kind of problem, you can pay attention to https:// docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_update_IDF.html

AntonDrubetskiy, 2021-09-10:
@Luxin812 Thanks a lot!))
v4.3 is the latest release branch. Which release branch would you recommend upgrading to in order for my project to work?

Luxin812, 2021-09-13:
The release version of ESP-AT based on the ESP32C3 platform and containing BLE commands is still in preparation. You can test and develop your project based on the ESP-AT master version. Each ESP-AT version has a compatible IDF version. If you want to upgrade the IDF version, you can refer to the document: https:// docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_update_IDF.html

AntonDrubetskiy, 2021-09-13:
@Luxin812 Thank you very much for the information!)))

AntonDrubetskiy, 2021-09-13:
@Luxin812 Another question. I know that this is not on the topic of this public. Tell me, is it possible to create your own custom service and characteristics using AT commands? There are such examples for Arduino, but I need it with the help of AT commands.
I need to create a UUID for a service and characteristic, etc. The examples given in the official documentation do not say anything about this, and the services and characteristics are already written in the file "example.csv"

Luxin812, 2021-09-14:
@AntonDrubetskiy Regarding how AT customizes the ble service, you can refer to this document: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_customize_BLE_services.html