espressif/esp-at #564: 自定义AT指令出错

AHulyi, 2021-09-10:
我参考文档docs/en/Compile_and_Develop/How_to_add_user-defined_AT_commands.rst，自定义一个发送数据的AT指令，
AT+IOTSEND=3
>
abc

OK
发送定长数据，代码如下：
uint8_t at_setup_iotSEND(uint8_t para_num)
{
    int16_t buf_len = 0;
    uint32_t received_len = 0;
    int32_t ret = -1;
    uint8_t *buf = NULL;

    if (ESP_AT_PARA_PARSE_RESULT_OK != esp_at_get_para_as_digit(0, &buf_len))
    {
        return ESP_AT_RESULT_CODE_ERROR;
    }

    buf = (uint8_t *)malloc(QIOT_AT_BUFFER_MAX);

    esp_at_port_write_data((uint8_t *)">", strlen(">"));

    while (buf_len != received_len)
    {
        received_len = esp_at_port_read_data(buf, buf_len);
    }

    ret = iotCmdBusPassTransSend(mode, buf, buf_len);//发送数据，成功返回1

    if (buf)
    {
        HAL_FREE(buf);
    }

    if (1 != ret)
    {
        return ESP_AT_RESULT_CODE_ERROR;
    }

    return ESP_AT_RESULT_CODE_OK;
}

串口打印：
[11:47:01.329]发→◇AT+IOTSEND=1,3
□
[11:47:01.335]收←◆AT+IOTSEND=1,3
>
[11:47:03.713]发→◇123

□
[11:47:03.718]收←◆
OK

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

[11:47:03.772]收←◆+IOTEVT: 4,10200

我想问的是为什么会有ERROR，我也按照例子中使用信号量的方法进行测试，发现一样有这个问题，请问这个要怎么解决？

AHulyi, 2021-09-10:
我又测试了例子Input Data of Specified Length
AT+TEST=<param_1>
发现是一样的问题

[13:44:36.640]发→◇AT+TEST=3
□
[13:44:36.645]收←◆AT+TEST=3
>
[13:44:37.855]发→◇123

□
[13:44:37.862]收←◆

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

received data is: 123
OK

ustccw, 2021-09-10:
@AHulyi 请提供下完整的测试代码和测试步骤。

AHulyi, 2021-09-10:
static xSemaphoreHandle at_sync_sema = NULL;

void wait_data_callback(void)
{
    xSemaphoreGive(at_sync_sema);
}

uint8_t at_setup_cmd_test(uint8_t para_num)
{
    int32_t specified_len = 0;
    int32_t received_len = 0;
    int32_t remain_len = 0;
    uint8_t *buf = NULL;
    uint8_t buffer[64] = {0};

    if (esp_at_get_para_as_digit(0, &specified_len) != ESP_AT_PARA_PARSE_RESULT_OK) {
        return ESP_AT_RESULT_CODE_ERROR;
    }

    buf = (uint8_t *)malloc(specified_len);
    if (buf == NULL) {
        memset(buffer, 0, 64);
        snprintf((char *)buffer, 64, "malloc failed\r\n");
        esp_at_port_write_data(buffer, strlen((char *)buffer));
    }

    // sample code
    // users don't have to create semaphores here
    if (!at_sync_sema) {
        at_sync_sema = xSemaphoreCreateBinary();
        assert(at_sync_sema != NULL);
    }

    // output input prompt ">"
    esp_at_port_write_data((uint8_t *)">", strlen(">"));

    // set the callback function which will be called by AT port after receiving the input data
    esp_at_port_enter_specific(wait_data_callback);

    // receie input data
    while(xSemaphoreTake(at_sync_sema, portMAX_DELAY)) {
        received_len += esp_at_port_read_data(buf + received_len, specified_len - received_len);

        if (specified_len == received_len) {
            esp_at_port_exit_specific();

            // get the length of the remaining input data
            remain_len = esp_at_port_get_data_length();
            if (remain_len > 0) {
                // sample code
                // if the remaining data length > 0, the actual input data length is greater than the specified received data length
                // users can customize the operation to process the remaining data
                // here is just a simple print out of the remaining data
                esp_at_port_recv_data_notify(remain_len, portMAX_DELAY);
            }

            // sample code
            // output received data
            memset(buffer, 0, 64);
            snprintf((char *)buffer, 64, "\r\nreceived data is: ");
            esp_at_port_write_data(buffer, strlen((char *)buffer));

            esp_at_port_write_data(buf, specified_len);

            break;
        }
    }

    free(buf);

    return ESP_AT_RESULT_CODE_OK;
}

static esp_at_cmd_struct at_custom_cmd[] = {
    {"+TEST", at_test_cmd_test, at_query_cmd_test, at_setup_cmd_test, at_exe_cmd_test},
};

esp_at_custom_cmd_array_regist(at_custom_cmd, sizeof(at_custom_cmd) / sizeof(at_custom_cmd[0]));//把这个放在app_main()中，文件中的数据内容按照SafeFlashHeadData_t结构体规定的内容格式

先发送：AT+TEST=3
出现 > 后
再发送 123

这个完全是按照文档docs/en/Compile_and_Develop/How_to_add_user-defined_AT_commands.rst中复制的代码

我使用的esp-at版本是release/v2.2.0.0_esp32

ustccw, 2021-09-13:
@AHulyi 代码是 okey 的，建议抓下你的线上数据，另外再 double check 下，是否有可能多输入了，不止 3 个字节的数据。

[image]

AHulyi, 2021-09-13:
找到问题了，我使用串口工具发送AT指令和数据，发送AT指令时，勾选了"加回车换行“这个选项，在 > 后输入的数据时也加了这个选项，就会出现如上错误，在发送数据时，不加回车换行，没有问题。

ustccw, 2021-09-13:
好的。多输入的数据会被 ESP-AT 当做下一条命令，因此会返回 ERROR

AHulyi, 2021-09-14:
你好，请问在esp-at中可以自定义蓝牙服务吗，我尝试编译gatts_demo.c编译失败，这个需要在menuconfig中配置些什么吗？

closed 2021-09-28 by xcguang