espressif/esp-at #566: ESP8266 MQTTCONN  exception

edieruby, 2021-09-16:
ESP8266
fw: v2.2.0.0  or v2.2..1.0
send MQTTCONN , response : ready
[image]

edieruby, 2021-09-17:
AT+GMR
AT version:2.1.0.0(ded5b14 - Jul 24 2020 12:01:05)
SDK version:v3.3-2-gb6e861a
compile time(36bb200):Jul 28 2020 08:30:34
Bin version:2.1.0(WROOM-02)

edieruby, 2021-09-17:
AT+MQTTCONN , the module restart.

[09:46:40.326] all parameters are ready
[09:46:40.375] Stack canary watchpoint triggered (mqtt_task)

Backtrace: 0x40275dca:0x3ffec5f0 Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 0 register dump:
PC      : 0x402202ad  PS      : 0x00000033  A0      : 0x40220220  A1      : 0x3ffe8df0  
A2      : 0x00065dba  A3      : 0x00000142  A4      : 0x3ffec633  A5      : 0x3ffec62f  
A6      : 0x000007fe  A7      : 0x000000a3  A8      : 0x000000c0  A9      : 0x00000000  
A10     : 0x3ffe8e30  A11     : 0x000000ff  A12     : 0x40275dca  A13     : 0x3ffe8e30  
A14     : 0x40276cc5  A15     : 0x40276cc5  SAR     : 0x0000001f  EXCCAUSE: 0x00000009  

Backtrace: 0x402202ad:0x3ffe8df0 0x40216aa7:0x3ffe8e20 0x40100eec:0x3ffe8e70 0x40220220:0x3ffe8e90 

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
[09:46:40.453] load 0x40100000, len 6960, room 16 
[09:46:40.466] tail 0
chksum 0x4f
load 0x3ffe8008, len 24, room 8 
tail 0
chksum 0xc6
load 0x3ffe8020, len 3196, room 8 
tail 4
chksum 0x3a
csum 0x3a+U礥獏TサZ昒*PUUUUō+V
磚Z┄Uア玌UP猅B﹗X暘i猆PY嫪猅Uア玌UP猅B┑ZUTU琔T　︰T玌UオTUUUTU+UP　I獼*UUオTUUjU╒UTZ匲U礥UJUオTUU猀UU╒Qù暘*RUZ┄Uア玌UP￡RJP　　Z元　　—X
U狹UZJPVエ┄Uア玌UPZUB﹗　　　ˉⅸ敳　　Q
QQQEUオTUU]Z寓　　　アe,P　　Q(QQQQQ┄Uア玌UPb獴┑Z(隦V碪P　知礫　　E(QQQEQQQEQ┄Uア玌UPj狟┑Z++　　　アE,P　　Q*QQUQUオTUU猀玓　　　—Rⅰ　ˉQQQQQ┄Uア玌UPJTUó叓P畢ZTT猆UE┄UU*UサZ瓍URkZU猇J～赫VMQ┄Uア玌UPJ擴ㄕ猇玍U礪Q睌T玅(Q奣Q禂T玅TU奣Q礥獘琓j橯QUX┄Uア玌UPJ誙ㄕ猇玍U礪睌T玅(QJR嫧ⅴ璔TUJRY礥獘珽U㏎瓡UB猉┄UUUPJUU—XV玍U礪U睌T玅(Q礥R禂T玅eQ礥獘,QQ(カ狟獣T┄Uア玌UPJ*U—XV玍U礪U睌T玅(Q奴浓ⅴ璔TQ～Z猀(QQQ(—珺獣T┄Uア玌UPJ猆ㄕ猇玍U礪睌T玅(Qe獷áキQT薄猀ㄕUQT┄Uア玌UPJUUóTJPXUkT磿ù]UZJP箠Q┄Uphy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
[09:46:40.715] module_name:WROOM-02
max tx power=78,ret=0
2.2.1
[09:46:40.867] [0;32mI (469) wifi:state: 0 -> 2 (b0)
[0m[0;32mI (500) wifi:state: 2 -> 3 (0)
[0m[0;32mI (505) wifi:state: 3 -> 5 (10)
[0m[0;32mI (598) wifi:connected with xxxxxx, aid = 7, channel 6, HT20, bssid = bc:1a:e4:c0:29:d8
[0m

ustccw, 2021-09-17:
@edieruby ESP8266 is not recommended as the preferred choice for MQTT connection over TLS, due to the relatively limited memory, and ESP32-C3 is recommended instead. You can use AT+SYSRAM? command to query the minimum available memory. 

According the log you provided, it seems to be a stack overflow issue.
For a test, you can try to enlarge mqtt_task stack:
`./build.py menuconfig`-> `Component config` -> `ESP-MQTT Configurations` -> `MQTT task stack size`.

edieruby, 2021-09-17:
[16:31:53.189] AT+SYSRAM?
+SYSRAM:45512,42160
OK

edieruby, 2021-09-18:
v2.1.0.0
[image]
[image]

ustccw, 2021-09-22:
same to https://github.com/espressif/esp-at/issues/568

edieruby, 2021-09-22:
not the same issue.
This means EPS8266 in V2.1.0.0 can connect to a MQTT broker by TCP over TLS ,
but ESP8266 in V2.2.0.0 or V2.2.1.0 can not connect to the same MQTT broker.

ustccw, 2021-10-21:
> > [09:46:40.326] all parameters are ready [09:46:40.375] Stack canary watchpoint triggered (mqtt_task)
> 
> I had the same issue when I was trying to do HTTPS requests. This error means that task stack overflowed because default stack size is too small for SSL. You should try going to menuconfig -> 'Component config' -> 'ESP-MQTT Configurations' -> 'MQTT task size' and increase it. My issue got resolved when I increased AT task stack to 4KB.

yes, i mentioned it at https://github.com/espressif/esp-at/issues/566#issuecomment-921581129.

ustccw, 2021-10-21:
All in all, ESP8266-AT should not used for MQTT over TLS, otherwise, the product is always with risk. User should consider the ESP32-C3 instead of ESP8266 in this case.

closed 2021-10-21 by edieruby