espressif/esp-at #568: MQTT connect exception

edieruby, 2021-09-18:
ESP8266
V2.1.0.0

ESP8266 connect to treafik:2.5.1 failed with tcp over tls.
get the error log:
[11:30:43.648] [0;31mE (611272) MQTT_CLIENT: Invalid MSG_TYPE response: 6, read_len: 0[0m

treafik:2.5.1 with tcp over tls  verified is OK.

ESP8266 connect to treafik:2.4.8 with tcp over tls is OK.

guess has some problems when build connection with treafik:2.5.1 over tcp over tls.

[image]
[image]
[image]

ustccw, 2021-09-22:
1. After connect failed, what is the output of `AT+SYSRAM?`
2. Have you tested whether MQTT over TCP connect to `docker xxx v2.5.1`

edieruby, 2021-09-23:
go version 1.17.x  optimize the TLS handshake.
I guess ESP8266 is OOM.

[image]
[image]

ustccw, 2021-10-21:
yes, you are right.

closed 2021-10-21 by edieruby