espressif/esp-at #571: 添加自定义AT指令问题

AHulyi, 2021-09-28:
我想在esp-at中添加自定义的AT指令，以Set Command:   AT+TEST=1,"espressif"  为例，
`uint8_t at_setup_cmd_test(uint8_t para_num)
{
    int32_t len = 0;
    uint8_t *str = NULL;
    uint8_t num_index = 0;
    uint8_t buffer[1024] = {0};

    if (esp_at_get_para_as_digit(num_index++, &len) != ESP_AT_PARA_PARSE_RESULT_OK) {
        return ESP_AT_RESULT_CODE_ERROR;
    }

    if (esp_at_get_para_as_str(num_index++, &str) != ESP_AT_PARA_PARSE_RESULT_OK) {
        return ESP_AT_RESULT_CODE_ERROR;
    }

    snprintf((char *)buffer, len, "%s\r\n", str);
    esp_at_port_write_data(buffer, strlen((char *)buffer));

    return ESP_AT_RESULT_CODE_OK;
}
static esp_at_cmd_struct at_custom_cmd[] = {
    {"+TEST", at_test_cmd_test, at_query_cmd_test, at_setup_cmd_test, at_exe_cmd_test},
};
esp_at_custom_cmd_array_regist(at_custom_cmd, sizeof(at_custom_cmd) / sizeof(at_custom_cmd[0]));
`
发送：
AT+TEST=241,"012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789"
我现在测试的结果是字符串的长度最大只能是240，请问这个大小可以更改吗？我要发送更长的数据该怎么做？

xcguang, 2021-09-28:
Please refer to https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_add_user-defined_AT_commands.html#input-data-of-unspecified-length

AHulyi, 2021-09-28:
你提供的这个方案我参考例程已经实现了，现在我就是需要一次输入所有的内容，是否有其他办法

ustccw, 2022-05-27:
i am sorry for that. `esp_at_get_para_as_str()` cannot parse a long string. Please refer to https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_add_user-defined_AT_commands.html#input-data-of-unspecified-length for that.

closed 2023-06-05 by ustccw