espressif/esp-at #572: Question on command AT + MQTTUSERCFG

AntonDrubetskiy, 2021-09-29:
Hello guys.
I need to connect to the AWS cloud over MQTT protocol.

I should use this command to do it:
AT+MQTTUSERCFG=<LinkID>,<scheme>,<"client_id">,<"username">,<"password">,<cert_key_ID>,<CA_ID>,<"path">

where:
LinkID - 0
scheme - 5
client_id - ""
username - ""
password - ""

My question is: if you need to connect to AWS you must provide also rootCA, client certificate and, private key. How should we add all these above-mentioned items?

ustccw, 2021-09-29:
@AntonDrubetskiy please take a look: https://blog.csdn.net/espressif/article/details/105045610?spm=1001.2014.3001.5501
There is a pretty simple way to config root CA and client certificate.

In fact, you also can generate a set of mqtt_ca.bin, mqtt_cert.bin, mqtt_key.bin by tools [ATPKI.py](https://github.com/espressif/esp-at/blob/master/tools/AtPKI.py), see [README.md](https://github.com/espressif/esp-at/blob/master/tools/README.md) for more details.
Once the certificate bin generated, you can use `AT+SYSFLASH` command or `AT+CIUPDATE` command to upgrade certificate bin what you want.

Besides, we are re-organizing this part of the document, thanks for your observation.

AntonDrubetskiy, 2021-09-30:
@ustccw thanks for the answer! Your link helped me!))

closed 2021-09-30 by AntonDrubetskiy