espressif/esp-at #580: esp32 esp-at ImportError: cannot import name 'PREVIEW_TARGETS'

zgzd0222, 2021-10-31:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:     ESP32-DevKitC
- Kit version DevKitC
- Module or chip used:  [ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    // v2.2.0.0_esp32c3-170-g283d237
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:    Linux
- Power Supply:         USB

## Problem Description

//Detailed problem description goes here.
when use the command: ./build.py menuconfig, the log as below:
from idf_py_actions.constants import GENERATORS, PREVIEW_TARGETS, SUPPORTED_TARGETS, URL_TO_DOC
ImportError: cannot import name 'PREVIEW_TARGETS'
idf.py build ret: 1
idf.py build failed

ustccw, 2021-11-16:
@zgzd0222 please provide all logs after you cloned esp-at project.

FBEZ, 2022-02-21:
I have a similar issue, but I get 

from idf_py_actions.constants import GENERATORS, PREVIEW_TARGETS, SUPPORTED_TARGETS, URL_TO_DOC
ImportError: cannot import name 'URL_TO_DOC' from 'idf_py_actions.constants' (C:\Users\Francesco\esp\esp-at\esp-idf\tools\idf_py_actions\constants.py)

## Log after cloning esp-at

C:\Users\Francesco\esp>git clone --recursive https://github.com/espressif/esp-at.git
Cloning into 'esp-at'...
remote: Enumerating objects: 12176, done.
remote: Counting objects: 100% (3950/3950), done.
remote: Compressing objects: 100% (1976/1976), done.
remote: Total 12176 (delta 2832), reused 2993 (delta 1932), pack-reused 8226 eceiving objects: 100% (12176/12176), 109.24 MiB | 8.93 MiB/s
Receiving objects: 100% (12176/12176), 111.74 MiB | 6.75 MiB/s, done.
Resolving deltas: 100% (7479/7479), done.

C:\Users\Francesco\esp>cd esp-at

C:\Users\Francesco\esp\esp-at>build.py menuconfig
Traceback (most recent call last):
  File "C:\Users\Francesco\esp\esp-at\build.py", line 25, in <module>
    import xlrd
ModuleNotFoundError: No module named 'xlrd'

C:\Users\Francesco\esp\esp-at>python -m pip install pyyaml xlrd
Requirement already satisfied: pyyaml in c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages (6.0)
Collecting xlrd
  Downloading xlrd-2.0.1-py2.py3-none-any.whl (96 kB)
     |████████████████████████████████| 96 kB 1.6 MB/s
Installing collected packages: xlrd
Successfully installed xlrd-2.0.1
WARNING: You are using pip version 20.3.3; however, version 22.0.3 is available.
You should consider upgrading via the 'C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.

C:\Users\Francesco\esp\esp-at>build.py menuconfig
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP32C3
choose(range[1,2]):2

Module name:
1. MINI-1 (description: TX:7 RX:6)
2. ESP32C3-SPI
3. ESP32C3_QCLOUD (description: QCLOUD TX:7 RX:6)
choose(range[1,3]):1

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):0
platform_name=ESP32C3,module_name=MINI-1
Please wait for the SDK download to finish...
Cloning into 'esp-idf'...
remote: Enumerating objects: 309424, done.
remote: Counting objects: 100% (192/192), done.
remote: Compressing objects: 100% (105/105), done.
remote: Total 309424 (delta 84), reused 188 (delta 80), pack-reused 309232
Receiving objects: 100% (309424/309424), 174.29 MiB | 6.22 MiB/s, done.
Resolving deltas: 100% (228786/228786), done.
Updating files: 100% (8422/8422), done.
old commit:c831d1f1e00891ab6b69db48b7889a19437ec237
checkout commit:87f8a0d5f11fea174f8a77bf230bb9caf6caf38e
Please wait for the update to complete, which will take some time
Already up to date.
Note: switching to '87f8a0d5f11fea174f8a77bf230bb9caf6caf38e'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

[... 92 lines cut ...]
Submodule path 'components/tinyusb/tinyusb': checked out '334e95fac52a607150157ae5199a19e11f843982'
Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'bb672b0437485fc7420add178299631692b15ac3'
Update completed
WARNING: IDF_PATH environment variable is set to c:\espressif\frameworks\esp-idf-v4.4 but idf.py path indicates IDF directory c:\users\francesco\esp\esp-at\esp-idf. Using the environment variable directory, but results may be unexpected...
Traceback (most recent call last):
  File "esp-idf\tools\idf.py", line 812, in <module>
    main()
  File "esp-idf\tools\idf.py", line 728, in main
    cli = init_cli(verbose_output=checks_output)
  File "esp-idf\tools\idf.py", line 686, in init_cli
    extensions.append((name, import_module(name)))
  File "importlib\__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "c:\espressif\frameworks\esp-idf-v4.4\tools\idf_py_actions\core_ext.py", line 13, in <module>
    from idf_py_actions.constants import GENERATORS, PREVIEW_TARGETS, SUPPORTED_TARGETS, URL_TO_DOC
ImportError: cannot import name 'URL_TO_DOC' from 'idf_py_actions.constants' (C:\Users\Francesco\esp\esp-at\esp-idf\tools\idf_py_actions\constants.py)
idf.py build ret: 1
idf.py build failed

FBEZ, 2022-03-07:
Any idea about what can cause this issue?

xcguang, 2022-03-17:
You can try:
1. unset IDF_PATH
or 
2. set IDF_PATH=c:\users\francesco\esp\esp-at\esp-idf

closed 2023-06-05 by ustccw