espressif/esp-at #581: ESP32 C3 SPI AT wrong response

luohaobin2022, 2021-11-03:
## Environment
master: ESP32_DevKitC_V4
slave: ESP32_C3_DevKitM-1 v1.0
- Development Kit:      [ESP32_DevKitC_V4|ESP32_C3_DevKitM-1 v1.0]
- Module or chip used:  [ESP32-WROOM-32E|ESP32-C3-MINI-1]
- IDF version (run ``git describe --tags`` to find it): 
   v2.2.0.0_esp32c3-152-g9e46e4e
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
  
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
I can not get the right response from slave when I use SPI AT
  for example:
  master send"AT\r\n"， slave responsed with "AT"

### Expected Behavior
master send "AT\r\n"， slave should response with"OK"
### Actual Behavior
master send"AT\r\n"， slave responsed with "AT"

takeyoshi, 2021-12-08:
Hi, 

I am trying to get the ESP32-C3 (specifically ESP32-C3-13U module from aliexpress) to work in SPI-mode too. The problem though is that the espressif documentation and the GitHub docs are either outdated and out of sync or I just don't get it.
I have compiled esp-at as instructed, but before that I chose SPI for transport because SDIO is not available anymore.

The logic-analyzer-images found at the bottom of "https://github.com/espressif/esp-at/tree/master/examples/at_spi_master/spi/esp32" show a different protocol than the SPI AT Workflow found at "https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_implement_SPI_AT.html". 
Everything is very confusing and the source code is not very readable either (https://github.com/espressif/esp-at/blob/master/components/spi_driver/at_spi_driver.c)....

The SPI-AT-Workflow suggests that the SPI-master must first "step 1: request to send" with the payload-size in the "data-info" section at the end of the send_request-command (0x01). The next step must be a wait for the handshake (IRQ-like) from the SPI-slave (ESP) and then the SPI-master must query the spi slave (0x02) "step 3: read slave status"...

If I follow this workflow sending "0x01 0x00 0x00 0x00 0x04 0x00 0xFE" (CMD, ADR, DUMMY, len byte-1, len byte-2, seq, magic), then waiting for a change on the handshake-line, then sending "0x02 0x04 0x00 0x00 0x00 0x00 0x00" (CMD, ADR, DUMMY, +4 bytes for slave_info response) then the ESP-slave 'might' fill the last bytes with "0x02 0x01 0xFC 0x0F" which I have difficulties to interpret too. These four bytes are supposed to be the "slave_state" and represent "len, seq, state" and the only way the above data would make sense would be "0xFFC (4092) 0x01 (seq no) 0x02 (rd/wr state)" and in the source code I saw that there's indeed a 4096 byte buffer, but the "state" is nowhere documented.

Also, the handshake-line (IRQ) is not well documented. I am monitoring the ESP right after reset using a logic analyzer. 
I have setup the handshake-pin with no pullup and no pulldown. After reset ESP pulls the handshake-line up on itself and then pulls it down to low, and after sending "0x01 0x00 0x00 0x00 0x04 0x00 0xFE" (4 bytes because I thought that "AT\r\n" requires 4 bytes) the handshake-line went up again and stayed there, no matter what I do.

Trying to read 4092 bytes all return 0xf6. Writing "AT\r\n" packed in a WR-command followed by WR-done does not work either.

Can someone tell me the correct workflow that would work? And how to interpret the handshake-line and the slave_state-response, and whether my data-info (4 bytes) are in the correct byte-order?

Thank you very much in advance.
Take

ustccw, 2022-05-27:
@luohaobin2022 @takeyoshi Sorry about the long delay in getting back to you. could you please try with latest version?

closed 2023-06-05 by ustccw