espressif/esp-at #589: AT+CIPRECVDATA mismatch in the number of bytes in the command and response

AntonDrubetskiy, 2021-11-29:
Hey!

I am using a TLS connection to connect to the server. In order to read the server's response, my application is waiting for a response (using passive mode of the receive mode of socket data). After receiving the message "+ IPD, <link ID>, <len>", my microcontroller sends the AT command AT + CIPRECVDATA = <link ID>, <len>. However, in the answer ESP32 for data on 29 less than in the command (<len> - 29).

For example:

```
+IPD,0,581

AT+CIPRECVDATA=0,581
+CIPRECVDATA:552,HTTP/1.1 200 OK
Server: nginx/1.10.2
Date: Mon, 29 Nov 2021 15:53:33 GMT
Content-Type: application/json; charset=utf-8
Content-Length: 100
Connection: keep-alive
X-Powered-By: Express
Access-Control-Allow-Origin: *
ETag: W/"64-ChmFkyw6+CLQKosKhkao3aL8SMw"
Set-Cookie: ly_session=s%3A3HdEAAFGzWwgT-0EBYrNR-l3PM3p35nC.XUgdyuZVfcml6Hdi6zwVbi7EWthAWPTYyUswzjtBj7c; Path=/; Expires=Wed, 08 Oct 2031 15:53:33 GMT; HttpOnly
Vary: Accept-Encoding

{"data":["https://xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"]}
OK
```
Can you tell me why this is happening?

jandrassy, 2021-11-29:
related https://github.com/espressif/esp-at/issues/223

ustccw, 2021-12-15:
@jandrassy @AntonDrubetskiy thanks for the report, will fix it.

ustccw, 2021-12-15:
@jandrassy @AntonDrubetskiy Sorry, we cannot make it from code side. The reason is that after established the ssl link, esp-at checks the link's some states or buffered length by socket fd. For example, if esp-at received a socket event which (actual buffered len = 0), we cannot directly check if the link closed or ssl has not received a completed segment data. 

we'll keep the behavior at the present, except there is a new elegant way.

closed 2022-03-24 by igrr