espressif/esp-at #594: 请问esp32c3 AT开发，ssl连接服务器的时候，证书只能烧录进esp32里才能使用吗

doudouladou, 2021-12-15:
请问esp32c3 AT开发，ssl连接服务器的时候，证书只能烧录进esp32里才能使用吗，有没有其他方法写入证书，如文件系统操作

ustccw, 2021-12-15:
@doudouladou you could use `AT+SYSFLASH` or `AT+CIUPDATE` commands to do it. More details are in [下载或者更新证书二进制文件¶](https://docs.espressif.com/projects/esp-at/zh_CN/latest/Compile_and_Develop/how_to_generate_pki_files.html#id7)

doudouladou, 2021-12-15:
谢谢你 ，我想再问个问题，如果是用AT+SYSFLASH命令去操作的话，频繁擦除会对模块的寿命有影响吗，或者说是擦写多少次会有问题

ustccw, 2021-12-16:
@doudouladou yes, generally speaking, the flash life of erase and write is about tens of thousands of times. If it exceeds the times, the flash operation will become longer or the data will become unavailable. please know.

doudouladou, 2021-12-16:
好的，谢谢

doudouladou, 2021-12-16:
[image]
我现在建立一个ssl连接，但他一直返回ERROR，我如何知道ERROR表示什么样的信息，比如：未连接成功，服务器拒绝了连接请求等详细信息。AT+SYSLOG=1似乎对我这个需求没有什么作用

doudouladou, 2021-12-16:
我现在的情况是，用sysflash写入了client_ca、client_cert和client_key三个bin文件，用AT+CIPSSLCCONF=3,0,0配置以后，使用AT+CIPSTART="SSL",<IP>,<PORT>,尝试建立ssl链接，但是返回error，我并不知道具体error在哪里
[image]

doudouladou, 2021-12-16:
我这边生成bin文件的方式如下：

doudouladou, 2021-12-16:
我这边生成bin文件的方式如下：

doudouladou, 2021-12-16:
我这边生成bin文件的方式如下
1：原本的三个证书文件如图所示**
[image]
**
2：通过更改文件后缀名，改为pem文件
[image]
3：用AtPKI.py将pem文件转为bin文件
[image]

4：用AT+SYSFLASH=0,"client_ca",0,<size> 命令擦除
5：用AT+SYSFLASH=1,"client_ca",0,<size> 命令写入

doudouladou, 2021-12-16:
不好意思，找到问题了，忘记同步网络时间了，现在可以连上了

doudouladou, 2021-12-16:
谢谢你@ustccw

ustccw, 2021-12-21:
@doudouladou no problem

closed 2021-12-24 by doudouladou