espressif/esp-at #596: Build for ESP8266 fails: 'cmake' must be available on the PATH

gigamegawatts, 2021-12-17:
## Environment

- Development Kit:      NodeMCU
- Kit version 
- Module or chip used:  WROOM-02
- IDF version (run ``git describe --tags`` to find it): 
    v3.4-22-g967752e2
- Build System:    Not sure - see below
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-lx106-elf-gcc (crosstool-NG esp-2020r3-49-gd5524c1) 8.4.0
- Operating System:     Linux
- Power Supply:      USB

## Problem Description

I'm trying to build esp-at for the first time. My target device is an ESP8266 NodeMCU with a WROOM module.  I'm following the instructions at https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/Compile_and_Develop/How_to_clone_project_and_compile_it.html#esp8266-series.  

Everything goes as expected until the end of step 3 (the first run of ./build.py menuconfig).  At the very end of the output, I see:

```
Update completed
'cmake' must be available on the PATH to use esp-idf/tools/idf.py
idf.py build ret: 2
idf.py build failed
```

I wasn't sure if that's expected, since I'm building for an ESP8266 which generally uses make, not cmake.  So, I carried on with step 4 and step 5.

At the end of step 5, I ran the ./build.py menuconfig command again, and I get:

```
~/esp/esp-at$ ./build.py menuconfig
module_name WROOM-02
platform_name=ESP8266,module_name=WROOM-02
'cmake' must be available on the PATH to use esp-idf/tools/idf.py
idf.py build ret: 2
idf.py build failed
```

I'm not sure if the build process for an ESP8266 is supposed to use cmake, or just make, so I don't know what the root problem is.

### Expected Behavior

The menuconfig page should appear

### Actual Behavior

As above:

```
~/esp/esp-at$ ./build.py menuconfig
module_name WROOM-02
platform_name=ESP8266,module_name=WROOM-02
'cmake' must be available on the PATH to use esp-idf/tools/idf.py
idf.py build ret: 2
idf.py build failed
```

### Steps to repropduce

I followed the instructions at https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/Compile_and_Develop/How_to_clone_project_and_compile_it.html#esp8266-series, as below:

1. (I already had ESP8266_RTOS_SDK v3.4 installed at ~/esp.  It works for other projects.)
2. ~/esp/$ git clone --recursive https://github.com/espressif/esp-at.git
3. ~/esp/$ cd esp-at
4. ~/esp/esp-at/$ git checkout release/v2.2.0.0_esp8266
5. ~/esp/esp-at/$ ./build.py menuconfig
6. I chose: PLATFORM_ESP8266, WROOM-02, and "no" to Silence mode. 
7. As mentioned above, I got a message "'cmake' must be available on the PATH to use esp-idf/tools/idf.py" at the end of the previous command, but decided to carry on anyway.  If that message shouldn't have appeared, then you can ignore the rest...
7. ~/esp/esp-at$ cd esp-idf
8. ~/esp/esp-at/esp-idf$ tar -xzf ~/downloads/xtensa-lx106-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz
9. ~/esp/esp-at/esp-idf$ git clone -b v3.4 --recursive https://github.com/espressif/ESP8266_RTOS_SDK.git
10. ~/esp/esp-at/esp-idf$ export PATH="$PATH:$HOME/esp/esp-at/esp-idf/xtensa-lx106-elf/bin"
11. ~/esp/esp-at/esp-idf$ export IDF_PATH="$HOME/esp/esp-at/esp-idf/ESP8266_RTOS_SDK"
12. ~/esp/esp-at/esp-idf$ python2.7 -m pip install --user -r $IDF_PATH/requirements.txt
13. ~/esp/esp-at$ ./build.py menuconfig
module_name WROOM-02
platform_name=ESP8266,module_name=WROOM-02
'cmake' must be available on the PATH to use esp-idf/tools/idf.py
idf.py build ret: 2
idf.py build failed

### Code to reproduce this issue

Not applicable.

## Debug Logs

Not applicable

## Other items if possible

Not applicable.

ustccw, 2021-12-21:
@gigamegawatts It seems that you have not completed this step: [First, set up the development environment according to ESP8266 RTOS SDK Get Started Guide v3.4.](https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/Compile_and_Develop/How_to_clone_project_and_compile_it.html#id1)
please know.

gigamegawatts, 2021-12-21:
Actually, the fix was just to install cmake  (e.g. apt install cmake).

Since cmake wasn't listed in the prerequisites, I thought it wasn't required when building for the ESP8266, or that I was supposed to use some version of cmake that was created by Espressif.

I'll close this issue.

gigamegawatts, 2021-12-21:
Fixed by installing cmake.

closed 2021-12-21 by gigamegawatts